bluerange-io/bluerange-mesh #105: Updating Over The Mesh

mabner1996, 2019-04-25:
Hello
I already managed to flash FruityMesh module inside several NRF 51822 chips.
However, every time we update the program, we need to manually flash it to all of the chips one by one. 
I want to ask if there is any way to update all the chips over the mesh?

In addition, I also find a product called BlueRange Mesh Gateway here https://www.bluerange.io/pricing/ 
Is that program capable of updating the chips over the mesh?

Thank you for your time and convenience,
Sincerely
Michael ABNER

volkrass, 2019-04-25:
@mabner1996 , OTA support over the mesh is part of the commercial license of Bluerange. Bluerange Mesh Gateway is a ready to use gateway for OTA update and connects your beacons to our Bluerange Platform. It is also possible to connect the Bluerange Mesh Gateway to your own cloud with additional effort. If you like you can send  your inquiry to sales@mwaysolutions.com

mabner1996, 2019-04-26:
Okay, I already sent an E-mail to that address, thanks

closed 2019-04-26 by mabner1996