bluerange-io/bluerange-mesh #107: "Send Queue is already full"

mabner1996, 2019-06-04:
Our program tried to send some lengthy data from sensors to FruityMesh Station Node.
Initially, the data is sent successfully. However, the station node stopped getting any new data after several iterations and the sensor node will print "Send Queue is already full" message on its terminal.

I want to ask if there is any solution to this problem? Thank you very much.

mariusheil, 2019-06-08:
Hello,

the message simply means that there is currently no space left in the send queue and that you have to retry later after some of the data has been sent. Message splitting is implemented in a way so that the message is sent to the next node and first reassembled before being sent to the next node. This introduces some latency into the network and is not a good way to send data if you want to transmit a lot of data. if you can, try to split the data into multiple chunks of 20 bytes yourself. We might implement another way of splitting the data in fruitymesh in the future so that it will only be reassembled on the receiving node. This would be rather easy if the data is reassembled on a gateway, but is rather complex to do on a node, as it is hard to have many reassembly buffers if the node receives spolit data from multiple nodes at once.

Does the problem persist if you wait some time? Because that should not happen.

mabner1996, 2019-06-13:
Hello, thanks for the answer.

We tried limiting the message length and the problem already solved.

closed 2019-06-13 by mabner1996