bluerange-io/bluerange-mesh #109: moving to newer sdk and newer softdevice?

romankc, 2019-07-10:
Are there plans to upgrade fruitymesh to use a more recent SDK and/or more recent softdevice?

mariusheil, 2019-07-10:
Hi,
we have it ported to SDK 15.2 internally, but it is not integrated into our master yet as we still have a lot of devices out there that are working fine with the old SoftDevice and we do not want to update them. We are planning to have the new SDK integrated next week. Then it will be possible to build against SDK11, 14 and 15.2 from the same codebase. We will probably also release that on github not so much later.

mariusheil, 2019-07-23:
Hi, just a quick update. Our programmer who does the porting had been on holiday for some time, so he is only starting now. So this will take some more days. We'll update this once we release a new version.

romankc, 2019-07-23:
Thanks for keeping both the software and your users updated.

mariusheil, 2019-09-08:
Hi, 

that took longer than expected, but we finally released the verseion which includes the SDK 15.2. Internally, we are only using SDK 15.2 for the nrf52840 chipset and still use the older SDK versions for the nrf51 and nrf52832, because we do not yet need the other features of the newer SoftDevices.

Marius

volkrass, 2019-09-11:
I would say we can close it. @romankc, if you disagree, please feel free to reopen it.

closed 2019-09-11 by volkrass