bluerange-io/bluerange-mesh #110: Building problems

LeoEras, 2019-07-17:
I have the following problems with the compiling part. I have donwloaded and configured all the required files described in the Quick Start and I wanted to build the project myself. But this is what the make tells me

Compiling file: nrf_drv_common.c
Compiling file: nrf_drv_gpiote.c
Compiling file: app_timer.c
Compiling file: ble_db_discovery.c
Compiling file: ble_radio_notification.c
Compiling file: RTT_Syscalls_GCC.c
Compiling file: SEGGER_RTT.c
Compiling file: system_nrf52.c
Compiling file: app_util_platform.c
Compiling file: nrf_section_iter.c
Compiling file: nrf_drv_saadc.c
Compiling file: nrf_saadc.c
Compiling file: app_scheduler.c
Compiling file: hardfault_handler_gcc.c
Compiling file: board_0.cpp
Compiling file: board_18.cpp
Compiling file: board_4.cpp
Compiling file: board_1.cpp
Compiling file: FruityMesh.cpp
Compiling file: Config.cpp
Compiling file: GlobalState.cpp
Compiling file: Main.cpp
Compiling file: Boardconfig.cpp
Compiling file: ScanController.cpp
Compiling file: GAPController.cpp
Compiling file: FruityHalNrf.cpp
Compiling file: GATTController.cpp
Compiling file: AdvertisingController.cpp
Compiling file: ResolverConnection.cpp
Compiling file: Node.cpp
Compiling file: ConnectionManager.cpp
Compiling file: MeshAccessConnection.cpp
Compiling file: MeshConnection.cpp
Compiling file: BaseConnection.cpp
Compiling file: AppConnection.cpp
Compiling file: ScanningModule.cpp
Compiling file: TestingModule.cpp
Compiling file: IoModule.cpp
Compiling file: StatusReporterModule.cpp
Compiling file: Module.cpp
Compiling file: EnrollmentModule.cpp
Compiling file: AdvertisingModule.cpp
Compiling file: DebugModule.cpp
Compiling file: MeshAccessModule.cpp
Compiling file: mini-printf.cpp
Compiling file: FlashStorage.cpp
Compiling file: Utility.cpp
Compiling file: Terminal.cpp
Compiling file: Logger.cpp
Compiling file: LedWrapper.cpp
Compiling file: RecordStorage.cpp
Compiling file: PacketQueue.cpp
Compiling file: gcc_startup_nrf52.s
Linking target: FruityMesh.out
C:\Users\DELL-2\AppData\Local\Temp\ccZVSuvf.ltrans0.ltrans.o: In function `ResetToDefaultConfiguration':
C:\Users\DELL-2\Desktop\fruitymesh/src/Boardconfig.cpp:112: undefined reference to `setFeaturesetConfiguration_dev(ModuleConfiguration*)'
C:\Users\DELL-2\AppData\Local\Temp\ccZVSuvf.ltrans0.ltrans.o: In function `Initialize':
C:\Users\DELL-2\Desktop\fruitymesh/src/Config.cpp:122: undefined reference to `setFeaturesetConfiguration_dev(ModuleConfiguration*)'
C:\Users\DELL-2\AppData\Local\Temp\ccZVSuvf.ltrans2.ltrans.o: In function `StatusReporterModule::ResetToDefaultConfiguration()':
C:\Users\DELL-2\Desktop\fruitymesh/src/modules/StatusReporterModule.cpp:89: undefined reference to `setFeaturesetConfiguration_dev(ModuleConfiguration*)'
C:\Users\DELL-2\AppData\Local\Temp\ccZVSuvf.ltrans2.ltrans.o: In function `IoModule::ResetToDefaultConfiguration()':
C:\Users\DELL-2\Desktop\fruitymesh/src/modules/IoModule.cpp:72: undefined reference to `setFeaturesetConfiguration_dev(ModuleConfiguration*)'
C:\Users\DELL-2\AppData\Local\Temp\ccZVSuvf.ltrans3.ltrans.o: In function `Node::ResetToDefaultConfiguration()':
C:\Users\DELL-2\Desktop\fruitymesh/src/mesh/Node.cpp:155: undefined reference to `setFeaturesetConfiguration_dev(ModuleConfiguration*)'
C:\Users\DELL-2\AppData\Local\Temp\ccZVSuvf.ltrans4.ltrans.o:C:\Users\DELL-2\Desktop\fruitymesh/src/modules/EnrollmentModule.cpp:87: more undefined references to `setFeaturesetConfiguration_dev(ModuleConfiguration*)' follow
collect2.exe: error: ld returned 1 exit status
Makefile:290: recipe for target '_build/release/NRF52/dev/FruityMesh.out' failed
make: *** [_build/release/NRF52/dev/FruityMesh.out] Error 1

I have no idea of what is going on, if I miss a file or if the issue is Windows 10. Thank you

mariusheil, 2019-07-18:
Hi,
it looks like you are trying to build the "dev" configuration. This is not available in the github version. The featureset that you are building for is configured in the Makefile at the top:

FEATURESET ?= github

All available featuresets are in the /config/featuresets folder. Maybe you set the FEATURESET variable somewhere in your environment profile. Try to use "Featureset = github" in the makefile (remove the ?). The do a "make clean" first, then make.

Good luck

LeoEras, 2019-07-18:
Thank you so much! It finally worked for me!

closed 2019-07-18 by LeoEras