bluerange-io/bluerange-mesh #111: Timer event handler questions

LeoEras, 2019-07-31:
Good evening,

I was testing the project and I came across the description of the timer in types.h
`//Returns true if the timer should have trigered the interval in the passedTime`
`#define SHOULD_IV_TRIGGER(timer, passedTime, interval) (interval != 0 && (((timer)-(passedTime)) % (interval) >= (timer) % (interval)))`
so I tried to put this function inside PingModule::TimerEventHandler(u16 passedTimeDs) to make a led blink every once in a while (like in IoModule). I think passedTimeDs is the second argument of SHOULD_IV_TRIGGER, but which are the first and the third one? I tried with many numbers but I see the same effect on a led: it either turns on immediately or it turns off immediately and stays that way until reset. I don't understand what is passedTimeDs either. Is it an accumulative amount? What are the arguments SHOULD_IV_TRIGGER takes, does it fire once? Thank you very much!

mariusheil, 2019-08-01:
Hi,

have a look at the status reporter module for how the arguments are used.
Timer is the variable that counts the total time while passedTimeDs is the
time that has passed since the last time the TimerEventHandler was called.
The third argument is the interval, e.g. 20 ds. All units are in deci
seconds, so using 20 as an interval and calling the led toggle function
should let the led blink every 2 seconds. You must however first set the
led mode to custom in the Io module, otherwise it will immediately
overwrite your led settings.

Hope that explanation works for you 🙂

LeoEras, 2019-08-01:
Yesm I set the led on custom mode. Thank you so much, it works perfectly now!

closed 2019-08-01 by LeoEras