bluerange-io/bluerange-mesh #112: System Off similar function

LeoEras, 2019-08-08:
Is there something akin to deep sleep to the fruitymesh? I know it's a weird question, but imagine I can get to deep sleep "sd_power_system_off()" on a custom board which can be woken up with a button event, then connect to the mesh to transmit data to sleep again. I've tried the DISCOVERY_OFF and every optimization I can, and so far have decreased power consumption until about 5uA, which is great for what I need, but I wonder if there's further optimization. Thank you.

mariusheil, 2019-08-09:
Hi,

we haven't yet had the use case of decreasing the consumption further. 5 uA
sounds about right if the chip is mostly sleeping and maybe occasionally
waking up with the 200ms timer. Without doing stuff like deactivating parts
of the ram, your best option is to use deep sleep. That's not yet
implemented, but you could just call the sofdevice directly and set it to
one of the sleep/off states. You would have to register a timer or some
kind of interrupt to wake up the chip again. But after powering on, you
will have lost your current state. If that does not matter, it is fine. It
will be like a reboot. Depending on how often you wake up, it is a better
idea to keep the mesh connections active and switch them to an interval of
4 seconds. Because setting up the mesh takes a lot of power and makes only
sense if you plan to sleep for a long time.

Marius

LeoEras, 2019-08-12:
Thank you, I'll try with that softdevice function

closed 2019-08-12 by LeoEras