bluerange-io/bluerange-mesh #115: Add Roadmap & Changelog to our README

volkrass, 2019-09-07:
To keep the community up-to-date, we should improve our README to give an overview what we are working on and what have been fixed with latest version (changelog).

deviousasti, 2020-03-22:
Is the NRF51 supported?
In `featuresets`, there's no nrf51, only nrf52 - it seems to have been removed in recent commits.

MichalBiolik, 2020-03-24:
Hi @deviousasti . NRF51 is depricated - it is still supported, but no new features will be added for NRF51 and it may come out of support soon. Regarding featureset - you can always create your own featureset targeted for NRF51 device.

deviousasti, 2020-03-25:
Thank you for the response.

mariusheil, 2020-09-21:
Closing this as it was inactive for some time, feel free to reopen.

closed 2020-09-21 by mariusheil