bluerange-io/bluerange-mesh #118: Unable to load custom modules

jjduhamel, 2019-09-26:
Since the recent update, I'm having trouble figuring out how to load and activate my custom modules.  I see the code related to this has changed, but the docs and example code are both still relevant to the old codebase.

In addition to the changes I made here: https://github.com/mwaylabs/fruitymesh/pull/117, I've made the following changes:

```
diff --git a/config/featuresets/github.cpp b/config/featuresets/github.cpp
index 933b665..8d981d4 100644
--- a/config/featuresets/github.cpp
+++ b/config/featuresets/github.cpp
@@ -37,6 +37,7 @@
 #include "EnrollmentModule.h"
 #include "IoModule.h"
 #include "MeshAccessModule.h"
+#include "PingModule.h"
 #include "GlobalState.h"
 
 void setFeaturesetConfiguration_github(ModuleConfiguration* config, void* module)
@@ -74,6 +75,7 @@ u32 initializeModules_github(bool createModule)
        size += GS->InitializeModule<EnrollmentModule>(createModule);
        size += GS->InitializeModule<IoModule>(createModule);
        size += GS->InitializeModule<MeshAccessModule>(createModule);
+       size += GS->InitializeModule<PingModule>(createModule);
        return size;
 }
```

Is there some other step that I'm missing?

Brotcrunsher, 2019-09-27:
Hi John!

You are right, the documentation for this is a bit outdated, a fix for it is already on the way. It seems like you have found out most of the new stuff by yourself. Probably the module is not active though. To do this you must set `configuration.moduleActive = true` for the PingModule. You can either do this inside of `PingModule::ResetToDefaultConfiguration()` or inside of `setFeaturesetConfiguration_github` in the case that `config->moduleId == ModuleId::PING_MODULE`. Hope this helps!

Jakob

BlockscapeLab, 2019-10-11:
So that you don't have to wait for the documentation to be updated for our github version. Here is a [link](https://www.bluerange.io/docs/fruitymesh/Implementing-a-Custom-Module.html) to the already released commercial version of the documentation. "Custom Modul" description is the same for commercial and non-commercial version.

volkrass, 2019-10-11:
So that you don't have to wait for the documentation to be updated for our github version. Here is a [link](https://www.bluerange.io/docs/fruitymesh/Implementing-a-Custom-Module.html) to the already released commercial version of the documentation. "Custom Modul" description is the same for commercial and non-commercial edition.

jjduhamel, 2019-10-14:
Thanks for the link.  I'm able to load the module, but when I issue `pingmod 0` or `pingmod this`, I see log messages from PingModule::TerminalCommandHandler but nothing from PingModule::MeshMessageReceivedHandler.

```
  < pingmod 0 
00> [PingModule.cpp@92 PINGMOD]: Trying to ping node 0 (201)
```

romankc, 2019-10-23:
@volkrass or @Brotcrunsher , any comments on this?

Brotcrunsher, 2019-10-28:
I have checked it and for me this is working. Could you give me more details please?

1. If you type in `status` you get a few details about the node. Could you tell me the `Mesh clusterSize` that is reported by this command? If both nodes are connected correctly it should report 2.
2. Did you enable the logging of the `PINGMOD` tag on both nodes?

Brotcrunsher, 2019-10-30:
I am happy that you were able to fix your issue. Yes, the module must be set active, as I wrote in my first response.

jjduhamel, 2019-11-01:
Ah yes I overlooked that.  Thanks for your help.

closed 2019-11-01 by jjduhamel