bluerange-io/bluerange-mesh #119: Issue with buttons on version 0.8.540

LeoEras, 2019-10-15:
Good morning!

I successfully migrated to version 0.8.5040 the code I was working on but I noted the button response both in the evaluation board and on a custom board doesn't work as expected. I now have to double press it to be detected. Everything is working as usual with commit  1bc7b3d where this isn't an issue. Could you please check this out?

MichalBiolik, 2019-10-16:
Hi,

I just tested and it looks fine for me. Are you testing default button used by mesh (Button 1) or did you implement your own button ?

LeoEras, 2019-10-16:
I was testing on Button 1 on nrf52 dk without changes and also in a custom board

MichalBiolik, 2019-10-17:
Can describe how you tested? What actions were you expecting? I added log in DebugModule in ButtonHandler function - it is printed every time button is pressed.

LeoEras, 2019-10-17:
I tested version 0.8.540 with the following changes in file FruityHalNrf.cpp for the button handler: 
```cpp
void button_interrupt_handler(nrf_drv_gpiote_pin_t pin, nrf_gpiote_polarity_t action){
	GS->ledGreen.Toggle();
        logt("ERROR", "btn pressed");

	//Because we don't know which state the button is in, we have to read it
	u32 state = nrf_gpio_pin_read(pin);
        ...
```
Nothing else was changed anywhere in the code. Clean and build fine without issues, but nothing happens if I press the button normally (as one press each second). 
The message "btn pressed" appears in pairs if I press the button repeatedly and fast (like 2 taps in half a second). The led also turns on and off immediately. This only happens this way.

MichalBiolik, 2019-10-23:
Sorry for late response.

I cannot confirm your problem. I edited `FruityHalNrf.cpp` and flashed `github` target on nrf52 dk. I see that everytime I press button `btn_pressed` is logged and LED toggles. Also the same happens when I release buttton. Unfortunately we don't have Rigado kit to check your setup.

LeoEras, 2019-10-23:
Ok, I understand. I have the system working without issues with previous version.

closed 2019-10-23 by LeoEras