bluerange-io/bluerange-mesh #122: How to send messages via UART?

NilsMinor, 2020-01-06:
Hi,

first of all I want to thank you for your awesome project!

I have a valid image running on a nrf51 evaluation board. When I reset the module I get valid informations via UART. But when I try to get information via command I do not get an answer from the module or I get 'Command not found'.
I tried the commands 
- get_info this
- get_modules this

Do I need to send an additional terminating character like \n? What am I doing wrong? ^^
Thanks,
Nils

NilsMinor, 2020-01-06:
Hi,
ok you can close my question, I found my mistake.
My serial tool added an CR+LF after the command. The correct option was to add only a CR.
Thanks anyway :D

mariusheil, 2020-01-06:
perfect :-)

closed 2020-01-06 by mariusheil