bluerange-io/bluerange-mesh #123: How to use a button with interrupt?

NilsMinor, 2020-01-14:
Hi,

what is the correct way to use button interrupts? I've seen that there is a callback "button_interrupt_handler". My Target is to register a callback function and to configure my button to call the routine. Should I do this inside my custom module?

My target is to add a PIR sensor to trigger functionality of my module.

Thanks,
Nils

mariusheil, 2020-01-15:
Hi,

look in Module.h and you will find a virtual ButtonHandler that you can override. This will be called once a button has been pressed and release for a certain time. If you need more functionality like e.g. starting as soon as the press occurs, you would have to implement this functionality yourself using the nordic methods. Otherwise, using our handler is the best approach. You can do this in your module.

mariusheil, 2020-01-16:
Hi,

what you could do to make it generic is to extend the functionality and call the ButtonHandler on Press and on Release events. Then, you need to pass the type of the event (pressed / released) and also the time (it is in deciseconds, like most other timeunits in fruitymesh, which is 1/10 second). For the pressed event, you could e.g. pass the time since the last released event. Not sure if somebody would be using that, but maybe better than just passing 0. If you do that, you could open a pull request and we would probably accept that.

Thanks

closed 2020-09-21 by mariusheil