bluerange-io/bluerange-mesh #124: Create a PWM Module?

NilsMinor, 2020-01-14:
Hi,

what is the correct way to create a PWM Module or is there already a PWM Module? I need to control three led with PWM. 

Should this be included inside my custom module? Or should I create a new Module with PWM functionality and make an PR to contribute to the project ? :)

Do you have any tips to use PWM on the nordic HAL to use PWM? I've used the nordic sdk 5 years ago, I think something has changed :D 

Thanks,
Nils

mariusheil, 2020-01-15:
We currently do not have a PWM module integrated. You should be able to use the one from nordic. They have libraries for Softblinking LEDs as well.
Best, if you currently do this in your own module. We are currently working a lot on our HAL which will then make it a lot easier to integrate nordic specific code in a generic way. You would be able to easily port this if you do it in your own module for now.

mariusheil, 2020-01-16:
> Hi,
> ok I will try to implement the Softblinking example. The configuration
> should be done inside my module initialization correct?
> When will the new HAL be released ?
> Thanks,
> Nils

It will be the same HAL, but we will have all nordic specific functionality
completely ported over and we will also be using interrupt based event
processing and the nordic softdevice handler which makes it easier to
integrate other nordic libraries. We will probably release parts of that in
1-2 weeks, but the rest might take something like 3-4 weeks i hope.

NilsMinor, 2020-01-20:
Hi,

I think I can wait until you updated to the new HAL in order to implement PWM. 
- Nils

NilsMinor, 2020-06-04:
HI @mariusheil,

currently I am trying to follow your explanation on how to implement ppm functionality.
I created my own class that implements the softlink led example 

```
       uint32_t err_code;
        uint32_t mask = (1 << inukPinConfig.lio1 | 1 << inukPinConfig.lio2 | 1 << inukPinConfig.lio3);
        const led_sb_init_params_t led_sb_init_param = LED_SB_INIT_DEFAULT_PARAMS(mask);

        err_code = led_softblink_init(&led_sb_init_param);
        logs("led_softblink_init %lu", err_code);
        APP_ERROR_CHECK(err_code);

        err_code = led_softblink_start(mask);
        logs("led_softblink_start %lu", err_code);
        APP_ERROR_CHECK(err_code);
```
I also set **#define LED_SOFTBLINK_ENABLED 1** to activate this module.

But I always get _undefined reference to `led_softblink_init'_. I have troubles on including the sdk15 specific _led_softblink.h_. Can you give me a tip what would be the correct way. Should I include files from sdk or is there an other "fruitymesh" way. Should I edit the CMake Files for this purpose? 

My goal is to fade 5 LEDs using PWM when an event triggered.
Thanks and best regards,
Nils

mariusheil, 2020-06-08:
Hi, can you give me the exact error that you get? Which platform do you compile for? Is it nrf52832 or nrf52840? We use SDK 14 for nrf52832 and only use SDk 15 for nrf52840 currently. You have to add files to cmake if you need that, that means you have to edit your custom featureset and add the led_softblink.c file. The #define should also be in the featureset.

NilsMinor, 2020-06-08:
HI @mariusheil thanks for your support!
I am using an nrf52832. I used code this to test :

```
#include <LioPWM.h>
#include "../../sdk/sdk14/components/libraries/led_softblink/led_softblink.h"

void LioPWM::InitLioPWM ( void ) {
    InukExtPins inukPinConfig;
	GS->boardconf.getSensorPins(&inukPinConfig);

    if (inukPinConfig.lio1 != -1) {
        uint32_t err_code;
        uint32_t mask = (1 << inukPinConfig.lio1 | 1 << inukPinConfig.lio2 | 1 << inukPinConfig.lio3);
        const led_sb_init_params_t led_sb_init_param = LED_SB_INIT_DEFAULT_PARAMS(mask);

        err_code = led_softblink_init(&led_sb_init_param);
        logs("led_softblink_init %lu", err_code);
        APP_ERROR_CHECK(err_code);

        err_code = led_softblink_start(mask);
        logs("led_softblink_start %lu", err_code);
        APP_ERROR_CHECK(err_code);
    }
    else {
        logs("No pwm pins found in GS");
    }
}
```
When compiling I get this output:

```
[100%] Linking CXX executable github_nrf52.out
/var/folders/99/b_vdv6n9545719gvh1pjrsvr0000gn/T//ccGwJqZ2.ltrans10.ltrans.o: In function `InitLioPWM':
/Users/nilsminor/Documents/inuit/inukmesh/src/hal/LioPWM.cpp:22: undefined reference to `led_softblink_init'
/Users/nilsminor/Documents/inuit/inukmesh/src/hal/LioPWM.cpp:26: undefined reference to `led_softblink_start'
collect2: error: ld returned 1 exit status
make[2]: *** [github_nrf52.out] Error 1
make[1]: *** [CMakeFiles/github_nrf52.dir/all] Error 2
```
I added the **#define LED_SOFTBLINK_ENABLED 1** to my featureset. The cmake to edit is : GitHub_nrf52.cmake?  Should I just add the led_softblink.c ? What about the include, I think this _../..sdk_ ... is not the right approach.

Thanks,
Nils

mariusheil, 2020-06-08:
Hi,

that error message helps, so it is a linker issue and not the issue that the header file was not found. This means that either the led_softblink.c file was not compiled at all (you can check that in the full output from the build process), or it means that it was compiled but e.g. some define was missing and that secion with the function was not compiled.

For now, you can edit the github_nrf52.cmake file and add the .c file, but it would be best to add a featureset for your custom implementation. I created a ticket so that we will add a tutorial in the feature on how to work with featuresets.

NilsMinor, 2020-06-13:
Hi @mariusheil,

I fixed the linker error by adding the following lines to **/sdk/sdk14/CMakeLists.txt**

```
target_sources_multi("${SDK_14_TARGETS}" "./components/libraries/led_softblink/led_softblink.c")
target_sources_multi("${SDK_14_TARGETS}" "./components/libraries/low_power_pwm/low_power_pwm.c")
target_include_directories_system_multi("${SDK_14_TARGETS}" "./components/libraries/led_softblink")
target_include_directories_system_multi("${SDK_14_TARGETS}" "./components/libraries/low_power_pwm")
```
Is this the correct way to add sdk specific sources?

I also got the _led_softblink_ example to work :) But i have some problems with it and hope you give me some advise as you know the nrf5x infrastructure better then me :D

The _led_softblink_ works periodic, but I need to control it like fade-on, stay-on and fade-out and stop (not starting again)._led_softblink_ uses the app timer to control the PWM duty cycle to fade the light smoothly. But it seams fruitymesh also uses this timer as when the LED fades and I connect or send via fruitymesh, the LED flickers. Any suggestions on how to build my led-fading software while there is fruitymesh communication ? I think to update it in fruitymesh periodic timer is to slow :/

Thanks :)
- Nils

mariusheil, 2020-06-15:
Hi,

the best way would be to add a new featureset (copy the three github* files from featuresets and name them according to your project. Then open the files and make sure you replace all the github* occurences with your own name, if there are any. Next, you should add the lines that you wrote above to your new featureset in the .cmake file. This way, it is possible to maintain multiple builds from the same source that includes different files depending on what the project needs.

As for the LEDs, you should probably disable the LED control in the IoModule in "your_custom_featureset.cpp". You can simply access the IoModule config and set the LED mode to CUSTOM (I believe). I do not think it is because of the timer but because of this. You have to make sure that fruitymesh uses a different timer than the led_softblink module, this is probably done during initialization or in the sdk_config for the softblink module.

Marius

closed 2020-09-21 by mariusheil