bluerange-io/bluerange-mesh #127: How to use the ADC correct?

NilsMinor, 2020-01-24:
Hi,

My application needs to use the ADC. I want to measure periodically (ever x minutes) the battery Voltage as well voltage from a solar panel.
I've seen you implemented battery measurement in the status reporter module. I think it makes sense to extend this module to add a second channel for my solar module or should I implement this in my custom module? 

How to use the add configuration? Can I just assign a ADC pin (e.g. P0.03 AIN4) to batteryAdcInputPin ? What is batteryMeasurementEnablePin good for? Is it used for lower power to cut off the add route? How to connect an 3.7 V lipo, should I use the internal resistors voltageDividerR1/R2 and is the max. voltage really 1.2V?

What is the better approach, should I send my battery information periodically or should I ask for the status through an external request? Like ask over an MQTT connector the mesh and a node(id) to send the current status?

Thanks for helping me, I relay like your project and I think it is pretty suitable for my application.
Nils

NilsMinor, 2020-01-28:
Hi @mariusheil any tips for me? :)

mariusheil, 2020-01-29:
Hi,

I think that is sth. you should do in your own module. We implemented that only to check the battery voltage of the node. I would suggest that you just disable our battery checking mechanism by setting the pins to -1 in your boardconfig for your custom board. Then you are free to do whatever you want. A collegue has written the ADC measurement, I would suggest that you check in the nordic documentation how the measurement should be done and how you can connect your battery and solar module. (Sorry for not being a great help here ;-))

As for the other part, it depends a bit on your use-case. If you want to monitor the power I think it is best to send that data every x minutes or so from the node, that is simpler than asking. Also you could implement sth. that only sends it when it has changed by some threshold, etc,...

The best approach to send your data would be by using component_sense messages. You should use your custom module id and then define the components and registers in your module. E.g. component could be SOLAR_PANEL_1, LIPO_BATTERY, ... and register could be BATTERY_VOLTAGE_VOLTS or sth. like that.
We do it in the same manner with our gateway and our platform has inbuilt historization and graphs for all data coming in as component_sense messages.

NilsMinor, 2020-01-30:
Hi,

thanks :) I will implement the ADC measurement like you described it and will include this inside my module so I can upgrade my project later when you push new features to master :)

I will try to use component_sense messages :)
Thanks,
Nils

NilsMinor, 2020-06-15:
Hi @mariusheil ,
i had time to test the ADC function using your HAL and it is working quite nice :) I used to read 1 ADC channel with it and implemented it the same way like the status reporter module. How can I add more channels? Do I have to set _AdcConfigureChannel_ before calling _AdcSample_ that will give me the adc value of the selected channel? Or can I set a change of channels that will be read out when called __AdcSample__ ? 

Thanks,
Nils

mariusheil, 2020-06-15:
Hi,

I do not have a lot of knowledge of the ADC module, so I will have to point you to the nordic devzone instead, sorry ;-)

Marius

closed 2020-09-21 by mariusheil