bluerange-io/bluerange-mesh #128: How to create a Gateway?

NilsMinor, 2020-01-26:
Hi,
I want to build a MeshGateway to connect my mesh to the internet. I thought on using an rpi with a serial connection to an nrf51/nrf52 module which acts as connection to the bluetooth mesh. The Pi runs a Qt application wich acts as MQTT client. What do you think of this setup?

How to build the gateway functionality into my nrf51/52 module. This module need to receive messages from the mesh and also needs to send messages to the mesh which it receives over serial connection. The gateway-node needs to know which nodes are in range and should handle the enrollment right? Should I use the meshAccessModule?

Any tips on architecture and software would help me a lot, thanks.
Nils

mariusheil, 2020-01-26:
Hi,
this is essentially almost what we are also doing. Most of the functionality you described is part of the open source release. You can use the terminal for sending in commands and getting the json responses from your "Communication Node". Create a new featureset and make sure to use the TerminalMode::JSON. You can then either use all the terminal commands or directly use rawsend to send binary data over the mesh. The enrollment can already be done over the mesh by issuing the appropriate terminal command of the enrollment module. Our Gateway node is directly part of the mesh and we do not use a mesh access connection. You can do that, but if you use an nrf51 or nrf52 you can just use the mesh firmware.

NilsMinor, 2020-01-26:
Hi Marius,
thanks again for helping me :)
Can I activate TerminalMode::JSON by simply define ACTIVATE_JSON_LOGGING 1 in my featrueset?
I think I will try to use serial commands as they are human readable. My first goal is to create one gateway-node and one normal node, but I still do not fully understand your concepts ^^.
I want to list all unenrolled nodes in my mesh und my gateway node and then I want to enroll them to be part of the mesh and to communicate with them over the gateway, but how?

Should I use the MeshMessageReceivedHandler to print received messages to my qt application?
How can I set a static nodeID or should I listen to ID 0 on my GateWay?
How can I send test data to this gateway from a node (for testing purpose).

Sorry for that many questions, I still did not fully understand the FruityMesh concepts but when I got this, I will hopefully  stop asking that many questions :D

Thanks,
Nils

NilsMinor, 2020-01-29:
Hi,
I think I understood how to enroll a node using the EnrollmentModule. But how can I scan for unenrolled nodes ?So are these steps correct?

1. The Gateway sets the network ??? using **action 0 enroll set_network**
2. The Gateway scans for unenrolled nodes, but how?
3. The Gateway enrolls the node using **action 0 enroll basic** to connect the node to the gateway mesh.

Is this the correct way to do it? ^^

mariusheil, 2020-01-29:
So many questions :-)

1)
Activating the JSON Terminal mode is done in your custom featureset under config/featuresets/yourFeatureset.cpp (take a look at the github template):
void setFeaturesetConfiguration_your_custom_featureset(ModuleConfiguration* config, void* module)
{
	if (config->moduleId == ModuleId::CONFIG)
	{
		Conf::getInstance().terminalMode = TerminalMode::JSON;
	}
}

2) There is nothing for listing the unenrolled nodes. You can try the "malog" command for starters. Our concept is, that all nodes have a QR code printed on them with the nodeKey and the serialNumber and you will have an App for enrolling them. You can also prepare a list of nodes with serial numbers and nodeKeys. Once you send the enroll command to a node, it will scan for a node with the given serial number and enroll it if it was found and if the nodeKey matches.

3) Yes, you can print received messages in the MeshMessageReceivedHandler, best to use some json output for that to be compatible. maybe some of the messages that you need (e.g. component_sense) are already printed by the gateway. make sure to set the gateway node to devicetype sink, e.g. in your featureset for the sink node (gateway node)

DeviceType getDeviceType_your_sink_featureset()
{
	return DeviceType::SINK;
}

4) Not sure what you mean, but a nodeId is usually given in the enroll command where you tell the node in which network it belongs and what nodeid it should use. Make sure you read the explanation in Specification.adoc for the nodeids.

5) There are some test commands, which will print some output if logging is enabled, e.g. "data". Or you can use component_sense
-------------------------
1) No, that's the wrong command, make sure you check the documentation : https://www.bluerange.io/docs/fruitymesh/EnrollmentModule.html

2) Try malog for starters, but you would not just look for all unenrolled nodes as you only want some specific nodes in your network most of the time, so you have to have some list of nodes you want to put in your network. Either prepare that manually or e.g. by using an App and QR codes.

3) You can use the live_reports, but you should start by using "action 0 status get_status" for the current status of all nodes or "action 0 status get_device_info" for getting the info that will only change after firmware updates or afte renrollment. Or "action 0 status get_connections" if you want to build a network graph or know who is connected to whom.
-------------
I think it is best if you read through all the documentation first. That will help you get a better understanding of everything. If there is sth. missing in the docs, we will try to add it.

Marius

NilsMinor, 2020-01-30:
Hi @mariusheil ,

thanks a lot for your support !

I will try it the way you mentioned and also will investigate your BlueRange documentation more to understand the backgrounds.

Today I got my second board and can now test much more :)
Nils

closed 2020-09-21 by mariusheil