bluerange-io/bluerange-mesh #129: Release 0.8.1700 changes

NilsMinor, 2020-02-11:
Hey,

I've seen you had released a new version of fruitymesh (thanks a lot!)
I tried to check what has changed but as 111 files changed, it's quite hard to figure out what exactly changed ^^
I've seen you changed a lot on the HAL layer and you created different github featuset. 
Is it still possible to use my nrf51 featureset?
Can you shortly describe what has changed or is the something like a release or version description?

Thanks,
Nils

Brotcrunsher, 2020-02-13:
Hi Nils,

since the 0.8.1700 release the support for nRF51 is deprecated. However, you can still use it, you just have to create your own [feature set](https://mwaylabs.github.io/fruitymesh/fruitymesh/Developers.html#Featuresets) for it. For a reference see config/featuresets/github_nrf52.cmake/.cpp/.h. For nRF51 we always used SDK 11, which is specified inside the .cmake file. You might run into some limitations that are no longer fixed by us (as I said, it's marked deprecated). That's the reason why I highly recommend upgrading to nRF52, if possible.

- Jakob

NilsMinor, 2020-02-17:
Hey @Brotcrunsher,
thanks for your response. I think I will do it as you advised it and will change to nrf52 hardware.
- Nils

closed 2020-02-17 by NilsMinor