bluerange-io/bluerange-mesh #131: Creating a Routing Table for FruityMesh

mabner1996, 2020-04-09:
In the "Concept" page of the wiki, under Relaying topic
It is stated that its possible to increase the throughput of the FruityMesh by using a routing table
Since normally the FruityMesh will create a connection automatically.

How does this work in the program and how do we implement a routing table in FruityMesh?
Do we need to specify which Node to connect 1st?

mariusheil, 2020-04-09:
Hi,

the idea for a routing table was more for routing in the existing mesh. It would be good if you could give us a short description of your use-case, e.g. why you need more throughput, if the nodes should form in a special pattern, etc... Then I could give you some better advise.

Thanks,
Marius

mabner1996, 2020-04-09:
Right now I just use the FruityMesh mesh to send data
But in the future, I might want to form topology like a line or a ring. The idea is that I can specify which node the data will be sent to in the codes

mariusheil, 2020-04-09:
Hi,
the topology is managed by the FruityMesh Algorithm to provide a self-healing network in all cases possible. If you remove a node, the connections will be set up so that the other nodes are reached through a different route. The way that nodes connect to each other is specified in the two methods CalculateClusterScoreAsMaster and CalculateClusterScoreAsSlave. It would be possible to change this method to provide a different strategy for telling the nodes to which nodes they should connect to. This will howevery require some digging through the code and the documentation to properly implement that. But you could use our simulator for implementing that feature as that will help you a lot: https://mwaylabs.github.io/fruitymesh/fruitymesh/CherrySim.html 
If you want to send data to some other nodes, you can check the Specification https://mwaylabs.github.io/fruitymesh/fruitymesh/Specification.html#NodeIds as there are a few possibilities to specify the recipients of messages. If you are interested to protect some data from some nodes, you could of course also chose to encrypt the data and only have the decryption key on some specific nodes.

Marius

mabner1996, 2020-04-20:
Noted. Thanks a lot for the advice

closed 2020-04-20 by mabner1996