bluerange-io/bluerange-mesh #132: Not able to Run the new Makefile to generate binaries

viveknath93, 2020-04-17:
Hi,
I had previously used fruity mesh and compiled the binaries in a different sequence of steps.
However, with the new fruity mesh, I am not able to compile the project.
Can you please help if I am missing something ? 

[image]
[image]

viveknath93, 2020-04-17:
Update: This is issue is solved after commenting out the following line from CMakeLists.txt
Not sure if this is a dirty workaround, but this works!
#  target_compile_options_multi("${NATIVE_TARGETS}" "-flto")

Also, the Docu for generating the Eclipse files can be updated from 
cmake ../fruitymesh -DBUILD_FEATURESETS=ON -DGCC_PATH="C:/path/to/gcc-arm-none-eabi-4_9" -G "Eclipse CDT4 - Unix Makefiles"

To
cmake ../fruitymesh -D BUILD_FEATURESETS=ON -D GCCPATH="C:\Program Files (x86)\GNU Tools ARM Embedded\4.9 2015q3" **-G "Eclipse CDT4 - Unix Makefiles"**

The older option is not working and it is already mentioned as a comment!

mariusheil, 2020-04-18:
Hi, thanks a lot for reporting this. We have had this problem on once PC running on OSX, but we were not sure why it is happening. We will look into this issue. This has something to do with the Hardfault handler that is implemented in the Nordic SDK.
We will look into this and see how that can be solved. Without -flto I suppose that the binary will be bigger. It is some compiler incompatibility that we did not yet figure out why it only happens on some PCs. I will report back.

viveknath93, 2020-04-20:
Hi,
Thanks a lot for the information.

mariusheil, 2020-04-21:
Hi,

I cannot reproduce the issue easily on my computer. I have checked that I have the same GCC version (yes, I do). Could you please do the following?:

- Create another build folder for cmake, next to your fruitymesh repository (e.g. fruitymesh_build_tmp and navigate to this directory.
- Use the following command (without modification) to build fruitymesh:
`cmake ../fruitymesh -DBUILD_FEATURESETS=ON -DGCC_PATH="C:/Program Files (x86)/GNU Tools ARM Embedded/4.9 2015q3" -G "Unix Makefiles"`

If an error occurs and you have to change the command, make sure to delete all files in the build directory first.

If this still leads to the same issue, can you please tell me which cmake version you are using? (cmake --version) (I have 3.16.3) and also which make utilities you are using (make --version) (I have: GNU Make 4.2.1 Built for x86_64-w64-mingw32).

Thanks

viveknath93, 2020-04-21:
Hi,
This seems to work fine now, I am able to generate the binaries now!

This is Myfolder structure:
C:\fruitymesh
   -\fruitymesh
   -\fruitymesh_build_tmp

Initially, The build directories were not getting created outside the repo folder. And were showing like "cache could not be found"
i.e : It was building only inside - C:\fruitymesh\fruitymesh
This was the reason why I was running build inside the repo folder itself.

I got the error mentioned above(screenshots) while compiling in this manner. and I commented out this below line from CMakeLists
**target_compile_options_multi("${NATIVE_TARGETS}" "-flto")**

cmake --version : cmake version 3.17.1
make --version : GNU Make 4.1 Built for i686-w64-mingw32

And yes. Hex files are smaller compared to previous compilation:
github_nrf52.hex -               351KB ->322KB
github_nrf52_merged.hex - 696KB ->667KB

Thank you!

closed 2020-04-22 by mariusheil

viveknath93, 2020-04-22:
thank you for the information!