bluerange-io/bluerange-mesh #133: Where is the implementation of "action 0 status get_connections" module?

mabner1996, 2020-04-22:
[image]

so if i type "action 0 status get_connections" in terminal, I will find a line 
"{"type":"connections","nodeId":9881,"module":3,"partners":[0,0,0,0],"rssiValues":[0,0,0,0]}"
the [partners] here shows all the nodeID of the nodes connected to it.
But I cannot find which line in the code is storing these partner NodeId data.

mariusheil, 2020-04-22:
Hi,

I'll just describe the full way of finding sth. like that for other people that will read this:

this is a command from the StatusReporterModule.cpp, you can do a full project search (Ctrl+Shift+F in most IDEs, sometimes Ctrl + H) for "get_connections". This will lead you to the line where the Terminal command is processed.

The message that is sent to request the connections is sent using the actionType "StatusModuleTriggerActionMessages::GET_ALL_CONNECTIONS". Looking for GET_ALL_CONNECTIONS in the whole project will lead to the line where the request is being processed and answered with: "StatusReporterModule::SendAllConnections".

Using Ctrl + Click on SendAllConnections will open the implementation of that method (somewhere around line 215 in my master branch in StatusReporterModule.cpp). This method is responsible for getting the partnerIds and copying it in a message that will be sent over the mesh. It does this by querying the ConnectionManager for the current connections: GS->cm.GetMeshConnections(ConnectionDirection::DIRECTION_IN);

This is then sent as a mesh message using the actionType StatusModuleActionResponseMessages::ALL_CONNECTIONS. Looking for this enum will lead us to the place where this response is processed and printed out to the terminal: (around line 630 with a comment: //Somebody reported its connections back).

Here, the json is constructed that is printed to the terminal simply by taking the message contents and putting the values into a string.

So the trick is to follow the enums through the code. The flow is mostly the same for most messages:
- Terminal Command being evaluated that sends a request message
- MeshMessageReceivedHandler (on the node that this message was directed to) being called and processing that request. In turn sending a response message
- MeshMessageReceivedHandler is called (On the node that the response was directed to) to process the response message.

mabner1996, 2020-04-22:
We found in statusreporter.cpp thats its actually from NodeMeasurements
thank you
{
		if(nodeMeasurements[i].nodeId != 0){
			NodeId sender = nodeMeasurements[i].nodeId;
			i8 rssi = (i8)(nodeMeasurements[i].rssiSum / nodeMeasurements[i].packetCount);

			memcpy(buffer + j*3 + 0, &sender, 2);
			memcpy(buffer + j*3 + 2, &rssi, 1);

			j++;
		}
	}

Thank you very much

closed 2020-04-22 by mabner1996

mariusheil, 2020-04-22:
Hi, but node_measurements is used for get_nearby and not for get_connections, right? You were asking for get_connections But glad that you found what you were looking for.