bluerange-io/bluerange-mesh #135: Create custom board configuration

NilsMinor, 2020-04-26:
Hey,

I designed a custom board (https://github.com/NilsMinor/inuk-base) based on the NRF52832. I have several GPIOs, UART, 2 adcs, and I want to create 5 PWM outputs. I have seen the board_config files but they are limited to the BoardConfiguration struct. What is the correct way to create a custom board config and to maintain the fruitymesh workflow?

Thanks and best regards,
Nils

mariusheil, 2020-04-27:
Hi,
as we cannot fit all different use-cases into this struct, we have decided to go with function pointers for  special use-cases. This is not yet in the current release and is not generic enough for everybody to be used. But I'll note it down, so that we can add it. In the meantime, you could do something like this:
In Boardconfig.h add:
`void (*getCustomPinset)(CustomPinset*) = nullptr;`
Then, in your custom board (e.g. board_123) where you configure the pins, you can add:
`GS->boardconf.getCustomPinset= &setCustomPinset_123;`
to pass the function pointer.
All you have to do then is to add that function to your custom board:
`extern void setCustomPinset_123(CustomPinset* pinsetConfig);

void setCustomPinset_123(CustomPinset* pinsetConfig){
	if(pinConfig->pinsetIdentifier == PinsetIdentifier::MY_PWM_SETTINGS){
		myPwmPins* pins = (myPwmPins*)pinConfig;
		pins->custompin1 = 7;
	}
}`
We have a struct that contains the pinsetIdentifier and the other structs with the custom settings inherit from this struct.
So all the generic stuff goes into the struct that we offer and all the more special stuff in the custom struct. We use a number of different pinsets for different sensors such as one for gyroscope, one for accelerometer, etc,....

Marius

NilsMinor, 2020-04-27:
Hi Marius,

thanks a lot, I will try it the way you described and will give you feedback if it worked for me :D

Thanks,
Nils

mariusheil, 2020-05-04:
Hi,

sorry for the late response, guess that slipped through.
You can just use the default board settings the way they are right now. Additionally you pass a function pointer to get your custom pins for PWM and other stuff.

Did you get it working?
Marius

NilsMinor, 2020-05-04:
Hi Marius,
thanks again for your support :)

Ok I got it, thanks. I added my bsp but without the function pointer but I will test this. I also need to build my hardware first :D

BR,
Nils

mariusheil, 2020-05-04:
Hi,

If you want some assembled PCB, you can check out JLPCB. I used that for a private project recently and had the assembled boards within less than 10 days. That's however a private comment and not something official from the company I am working for :-)

NilsMinor, 2020-05-04:
Hey,

that is exactly what I did :D. I always wanted to test JLC and I wanted to learn KiCad (switch from Altium). So I created a board (https://github.com/NilsMinor/inuk-base) and order SMD assembled PCBs :) But I still have to solder the remaining parts ^^.
However, I appreciate any tips :)

mariusheil, 2020-09-21:
Closing this as it was inactive for some time, feel free to reopen.

closed 2020-09-21 by mariusheil