bluerange-io/bluerange-mesh #136: Eclipse could not find types after cmake build

NilsMinor, 2020-04-26:
Hi, 

I created my project like described using cmake. The build worked fine and I created an hex file. But my Eclipse shows a lot of errors and (Type 'uint32_t' could not be resolved / Type 'BoardConfiguration' could not be resolved ...). I thought I should not make changed on my Eclipse setup as it was created through cmake? Is this due to a wrong setup or have I made some mistakes?

Thanks,
Nils

mariusheil, 2020-04-27:
Hi,
we are currently updating our documentation, have a look here for a more up-to-date version until we've updated the github version: https://www.bluerange.io/docs/fruitymesh/Quick-Start.html

We recommend using VsCode as the Eclipse indexer seems to have problems with cmake. The Cmake support seems to be outdated and poorly supported. There is also an eclipse plugin for cmake which you can try to use if you want to continue using eclipse. I have had a better experiance with that. But VSCode is really brilliant with Intellisense and also works on all platforms.

Marius

NilsMinor, 2020-04-27:
Hi Marius,

thanks I think I will change to visual code, thanks :) I use VS code more often than eclipse so I will try it that way.

Best regard,
Nils

closed 2020-04-27 by NilsMinor