bluerange-io/bluerange-mesh #137: How do I run CherrySim?

mabner1996, 2020-05-19:
Hello, I am trying to run CherrySim so I tried these steps as instructed in https://github.com/mwaylabs/fruitymesh/blob/master/docs/opensource/modules/ROOT/pages/CherrySim.adoc
and https://github.com/mwaylabs/fruitymesh/blob/master/docs/opensource/modules/ROOT/pages/BuildingWithCMake.adoc#creating-native-build-files-for-the-simulator
 but I found some problems which I am not quite sure how to follow

So far this is what I did:
1. I download the FruityMesh repository from this github as fruitymesh-master and I copied the same repository and rename the copy as fruitymesh
2. I downloaded and installed CMake and added the bin into my computer path
3. I downloaded Visual Studio 2017
4. I open developer command tool window Visual Studio 2017 and tried to run cmake ../fruitymesh in the fruitymesh directory

However the command is not succesfully and it shows some problem which I have screenshot
how do I resolve this?
[image]
[image]

mariusheil, 2020-05-19:
Hi,
you should not execute cmake within the repository itself. The documentation states "Create a sibling directory next to the repository". That means, you should create a directory in your FruityMesh2020_CHERRYSIM directory. This directory should be empty, and from this empty directory, you should execute the command. The other possibility is to create a folder in the repository "<fruitymesh>/_build/vs/" and in this directory execute "cmake ../../".

Can you try if this solves the issue?

mabner1996, 2020-05-20:
Ah I see, so right now I inside FruityMesh2020_CHERRYSIM directory I have two folder
1. fruitymesh-master : which is the git folder I've downloaded from this github
2. fruitymesh : which is the empty folder (sibling repository)

according to that, I tried to use terminal on fruitymesh folder directory and run the command:
cmake../fruitymesh-master

However I still get some error as seen in the attached screenshot
There is a VC++ project inside the sibling repository now which  I can open but it does not show anything
[image]

mariusheil, 2020-05-20:
Hi,

I am not sure what the issue could be. Could you make sure that you delete all CmakeCacke.txt and CMakeFiles directories (including contents) from the fruitymesh directory. Then also delete all files in you build directory (fruitymesh). Then try to execute the command again. Whenever there is an issue, you should always delete all build files.

Other than that, you could try googling a bit for the error "failed to run msbuild command to get the value of VCTargetsPath" as this might happen in some cases for some different reasons.

A coworker of mine will be back on friday and might be able to help some more.

mabner1996, 2020-06-03:
I tried to delete CmakeCacke.txt and CMakeFiles directories from the FM directory and the build files, but the same error still exist.
So actually before the error message there is a "Selecting Windows SDK version 10.0.17763.0 to target Windows 10.0.18363."
So is it possible because I don have the SDK for my windows (10.0.18363)?
And I cant find my version SDK from the Visual Studio 2017 installer

[image]

Brotcrunsher, 2020-06-04:
Hi! Unfortunately no one of our team has encountered that issue before, so all we can do at this point is to guess. You could try to pass the following argument when using CMake: `-A Win32`. We had issues with 64bit/32bit setups before so there is a slim chance that this is the real underlying issue here. Don't forget to remove all CMake generated files before executing CMake with the modified parameters! 

You could also try to install Visual Studio 2019 and try it with that.

mabner1996, 2020-06-05:
I tried to use that argument after removing all folder in my build directories, but I still can't built it properly even after using visual studio 2019 developer command prompt

Just to make sure, I need to empty my sibling directory before tried to use cmake again right? 
I didnt see CmakeCacke.txt and CMakeFiles directories (including contents) from the FruityMesh-master directory as the author mentioned before

Brotcrunsher, 2020-06-05:
Yes, the sibling directory should be completely empty. Are you sure that your Visual Studio is correctly installed? Are you able to compile other C++ projects with it?

mabner1996, 2020-06-07:
This is the first time I use visual studio actually, because previously I use FruityMesh from last year version( 2019) which does not use CMake and does not need visual studio

Brotcrunsher, 2020-06-08:
Okay, let us test your CMake and Compiler setup then. For this I prepared a simple hello world project for you here: https://github.com/Brotcrunsher/CMakeHelloWorld

Please download the Repo, create a sibling just as with FruityMesh, open a terminal in this sibling, and then execute:

    cmake ../CMakeHelloWorld
    cmake --build .

Please attach the logs.

mabner1996, 2020-06-09:
Thank you
I tried the same one, but it have the exact same error message as before

Here I attached the log as a txt file

mariusheil, 2020-06-09:
Hi,

I think you have attached the wrong log file. But it also looks like you have some setup issue on your computer that does not have anything to do with fruitymesh, because the example project was a plain-standard Hello World CMake project.
I guess we can't be of much more help here. I would suggest that you uninstall all Visual Studio Versions and CMake and then reinstall Visual Studio 2019 and the latest CMake version. Probably check your PATH as well after uninstalling everything to make sure there are no leftovers.
If this does not help, you probably have to check for some CMake Tutorials with Visual Studio and check out if they offer some troubleshooting.

Marius

BlockscapeLab, 2020-06-17:
@mariusheil, @Brotcrunsher 
Could we provide with a docker image to give developers a fast path to use CherrySim?

mariusheil, 2020-06-17:
Hi,

we actually have the buildbox that we are using for our pipeline on dockerhub (we are currently using version 19 in our pipeline):
https://hub.docker.com/r/relution/buildbox-cherrysim

All the tools for compilation are installed. Funnily, I have written the documentation on how to use this manually just yesterday, but so far this is not yet ready to be open sourced. Basically, the build instructions go like this:

- Check out the fruitymesh repository either in your docker container or map the repository in your docker container by using sth. like: `docker run -v /home/youruser/fruitymesh:/builds/relution/fruitymesh -it relution/buildbox-cherrysim:19`
- Next, in the interactive docker shell, go to the fruitymesh folder, start the cmake build and run the tester. The steps below are from our pipeline build, but I guess you could just follow the CMake documentation that we already have and execute the given commands in the buildbox.
`
mkdir -p ../cmake_build/cherrysim_tester 
    - pushd ../cmake_build/cherrysim_tester 
    - cmake ../../fruitymesh -DCI_PIPELINE=ON
    - cmake --build . --target cherrySim_tester
    - popd
    - ../cmake_build/cherrysim_tester/cherrySim_tester GitLab false
`

Tell me, if that helps you or if you ran into problems. I could incorporate these steps into our documentation.

Marius

mabner1996, 2020-07-07:
So I just recently managed to build the the CMake.
Actually I missed a lot of additional build tools I need to install through visual studio here: https://social.msdn.microsoft.com/Forums/sqlserver/en-US/4669c607-d718-44b8-81c6-1d421162fead/c-compiler-location?forum=visualstudiogeneral

In addition I also run vcvarsall.bat through the Visual Studio developer window for both 64 and 86 system

and then I can run the simulator by running the cmake in the sibling directory

Thank you so much for the help

closed 2020-07-07 by mabner1996