bluerange-io/bluerange-mesh #138: Usage of Utility::SaveModuleSettingsToFlash outside of Module::MeshMessageReceivedHandler

xvitus, 2020-05-20:
Hello,

We are using FM 0.8.540.

We need to update some configuration data (module settings) when the user pushes some buttons on the hardware.
So we need to invoke Utility::SaveModuleSettingsToFlash outside of MeshMessageReceivedHandler.
We are trying to update module settings (configuration) data inside TimerEventHandler, using the following code:

	SaveModuleConfigAction userData;
	userData.moduleId = moduleId;
	userData.sender = 0; //zero because we dont have packetHeader->sender outside of MeshMessageReceivedHandler
	userData.requestHandle = 0; //zero because we dont have packet->requestHandle outside of MeshMessageReceivedHandler

	Utility::SaveModuleSettingsToFlash(
		this,
		this->configurationPointer,
		this->configurationLength,
		this,
		(u8)ModuleSaveAction::SAVE_MODULE_CONFIG_ACTION,
		(u8*)&userData,
		sizeof(SaveModuleConfigAction));

But it seems that nothing is saved instead.
Can you please help suggesting the correct way to invoke saving outside of MeshMessageReceivedHandler?

Regards,
--Vito.

mariusheil, 2020-05-22:
Hi,

it would be helpful if you could provide a log of what happens. SaveModuleSettingsToFlash is just a utility function that internally calls a method from the RecordStorage class, which uses the FlashStorage. Make sure that the logtags "RS" and "FLASH" are enabled, then try to call the method and then attach your logfile. If you want to check yourself, you can also add a few logs in the methods to see what is being called.

xvitus, 2020-05-24:
Hi,

thank you, we followed your suggestions and found the issue was at our side.
Thank you very much for your help.

Regards,
--Vito.

closed 2020-05-24 by xvitus