bluerange-io/bluerange-mesh #140: Quickstart build fails

ojousima, 2020-06-25:
Hello,

I was following instructions at https://www.bluerange.io/docs/fruitymesh/Quick-Start.html and got error when compiling:
`Error: CFI instruction used without previous .cfi_startproc` After finding 
https://devzone.nordicsemi.com/f/nordic-q-a/31205/error-cfi-instruction-used-without-previous-cfi_startproc
I changed line 3524 of sdk_config.h to `#define HARDFAULT_HANDLER_ENABLED 0` and the project builds. 

Do I have a problem with my environment or should I open a pull request with that change to the sdk_config.h?

mariusheil, 2020-06-26:
Hi, I guess this is the same as this issue here that was solved:
https://github.com/mwaylabs/fruitymesh/issues/132

ojousima, 2020-06-26:
Yes, it seems to be same issue, I'm on OS X too. The [troubleshooting section](https://www.bluerange.io/docs/fruitymesh/VsCodeSetup.html#Troubleshooting) doesn't have documentation about this error though. Is removing the `-flto` or disabling hardfault handler a better workaround?

mariusheil, 2020-06-26:
Hi,

I believe that the issue was solved without modifying anything in FruityMesh. A clean build got everything working. As I remember a collegue of mine also had the problem on OSX but a clean build solved it. Try to follow these steps:
https://github.com/mwaylabs/fruitymesh/issues/132#issuecomment-617192558

ojousima, 2020-06-26:
Curiously, after running `cmake ../ -DBUILD_FEATURESETS=ON -DGCC_PATH="/opt/local/gcc-arm-none-eabi-4_9-2015q3/" -G "Unix Makefiles"` and compiling project with `/usr/local/bin/cmake --build /Users/otsojousimaa/git/fruitymesh/_build --config Debug --target all -- -j 10` the project links and I can't repeat the error in VS Code either no matter what I try. 

I'm closing this issue as I can't repeat it and you have a fix for it. Thanks for the help :)

closed 2020-06-26 by ojousima