bluerange-io/bluerange-mesh #141: Cherrysim, I cannot change the number of the nodes

mabner1996, 2020-07-07:
Hello, right now I can build and run the cherrySim_runner.exe
and I have 8 nodes shown in the localhost webserver
However, I tried to make a simulation with more nodes with command "sim nodes 100"
and the exe does not restart and the terminal does not do anything.
According to https://mwaylabs.github.io/fruitymesh/fruitymesh/CherrySim.html
I should be able to restart the simulation with 100 nodes if I type "sim nodes 100" in the runner terminal.
So how does this command actually
 work?
[image]

note: I attached the screenshot of the terminal when I entered "sim nodes 100" command

Thank you

mariusheil, 2020-07-07:
Hi,

looks like you have a link to our old documentation. We need to clean up a bit here. I will remove that link from the wiki page. How did you find that page? We moved our documentation to:
https://www.bluerange.io/docs/fruitymesh/CherrySim.html#_general_2

There was also a small bug in the simulator so that it was not returning "Command not found", we will fix this for the next release.

You have to give the featureset as well: "sim nodes 100 github_nrf52", this will set the number of nodes for this featureset. This allows you to simulate different node types in the same mesh (You can also enroll different nodes in different networks at the same time, which is quite neat :-))

Marius

mabner1996, 2020-07-09:
Thanks now I can make simulation of for different number of nodes.
I have several other questions:

1. So what are the arguments available for featureset of sim nodes function

2. I want to try to use cherrysim to simulate several node formation in FruityMesh
However I seem to not able to change the position of the nodes. 
I tried to use all the set_position and add_position but the runner seems to...stop running
I provide the screenshot down below to see, which I tried to add position to node BBBB1
[image]

Brotcrunsher, 2020-07-10:
Hi Mabner!

1. By default only "github_nrf52" is available, however you can create your own custom featuresets. These are located in `fruitymesh/config/featuresets`. To add a new featureset you can copy the `.cpp`, `.h`, and `.cmake` files, rename them, change the function names in the `.cpp` file (e.g. `setBoardConfiguration_X` where X is your featureset name) and add the name to `CherrySim::PrepareSimulatedFeatureSets()`. Some more info about featuresets can be found here: https://www.bluerange.io/docs/fruitymesh/Developers.html#Featuresets
2. There indeed seems to be a bug in our github source release. This does not happen with our internal repository and thus was not spotted until now. I'll have a look at it and get back to you later.

Brotcrunsher, 2020-07-10:
Hi again!

The issue has been fixed in the new Release 0.8.3940.

mabner1996, 2020-07-16:
Okay then, thanks a lot

closed 2020-07-16 by mabner1996