bluerange-io/bluerange-mesh #143: Saving and Loading of CherrySim

mabner1996, 2020-07-16:
Does CherrySim allows saving/loading of nodes configuration? 
Every time I open CherrySim runner it always use default configuration (8 Nodes, randomly generated)

I want to simulate a specific location of the nodes with CherrySim and save it so everytime I open the runner I can immediately go to the setting I already made previously.

Thank you for your time and consideration

mariusheil, 2020-07-16:
Hi,

yes, this is possible. You have multiple options:

- You can write a test and use cherrysim_tester to execute this test using a Gtest filter in CherrySimTester.cpp, e.g.:
`::testing::GTEST_FLAG(filter) = "*YourTestName*";`

This test can either use the device import from json (see https://github.com/mwaylabs/fruitymesh/blob/master/cherrysim/test/TestClustering.cpp => DoClusteringTestImportedFromJson) or you can use fixed positions that you give in the code (see TestEmergencyDisconnect => preDefinedPositions)

You can also specify the json files on the commandline using the following commands in CherrySimRunner:
`sim site C:/projects/fruitymesh/cherrysim/test/res/example/site.json
sim devices C:/projects/fruitymesh/cherrysim/test/res/example/devices.json
sim json 1`

I have attached two exemplary json files for you:
[example.zip](https://github.com/mwaylabs/fruitymesh/files/4930179/example.zip)

I guess we will add these example jsons to a future release.

Marius

mabner1996, 2020-07-23:
thank you very much

Is there any guidance/references for the Json files so I might know which options are available for writing the Json code

For example, in the Json code example you give me, the position of x,y,z of each nodes are actually percentage of the width and height of the site map, is it possible for the x,y,z coordinate to be the actual coordinate in meters in the actual map?

mariusheil, 2020-07-23:
Hi,

there is no real documentation at the moment, sorry. Best, if you check the code part that does the parsing. It is kind of trivial and you have to give the coordinates as relative positions from 0 to 1. You can e.g. use a site with 100 by 100 meter. Then you can simply give the coordinates in meter and divide them by 100 e.g.

Marius

mabner1996, 2020-07-26:
Ok thank you
last question I want to ask is that:
Is cherrysim behave according to the Fruitymesh directory when it was build?

So if for example, I change the clusterscore algorithm in the node.cpp and then rebuild cherrysim again
Is the change I made in the program be reflected in the Cherrysim?

Thank you so much

mariusheil, 2020-07-26:
Hi,

yes. CherrySim uses as much of the FruityMesh implementation as possible. So it will be almost the exact same behaviour that you will experience with real nodes.
You can always add a log statement to the FruityMesh code to be really sure that the change you made is correctly executed in Cherrysim.

Marius

mabner1996, 2020-07-28:
Okay then, thank you very much for your explanation!

closed 2020-07-28 by mabner1996