bluerange-io/bluerange-mesh #145: Question: One Mesh, Two Segments and a serial bridge

AntonWert, 2020-07-28:
Hi,
I have running 2 separate meshes, but I want them to combine. They should idealy behave as one big mesh sharing all data.
The connection should be trough wires to bridge some non-radio friendly environment.
Is this possible to send out data on one node to a serial connection and insert it on another node?

mariusheil, 2020-07-28:
Hi, I do not think that we have this prepared currently. you could probably bridge them by using the standard terminal. You could look for the place where data is received and routed. If you enable the logtag conn_data, you should see some logs with the binary data being received. Then you could use the raw data command to insert into the terminal of the other node. You will have to make sure that there is no clash between the nodeIds in the two networks. You might need to dig through the data routing function a bit to only relay the proper packets, but there are many comments that should make it easy to understand.

Marius

AntonWert, 2020-07-29:
Thanks Marius for sharing the idea.
I doubt that handling two parts of a mesh is ideal in case of message translation and the hint with the idea checks you described.
Just looking on the shown items I think about another solution: a node is having at least two types of (mesh) connections: a single connection to another master and several incoming connections – right?
If this would be correct, isn’t it possible to have the two parts of the connection on two µC’s and share the data on a serial line? Like splitting the ConnectionManager into to devices.
What do you think?
Anton

mariusheil, 2020-08-06:
A node will always have one connection, not necessarily specified if this is a central or peripheral connection. But usually a node can have a single meshconnection where it is the peripheral and can have up to 3 connections to others where it is acting as a central. You could certainly implement a new connection that subclasses the BaseConnection and you could modify the connectionmanager to also route to this connection of which you will e.g. always have one instance.

mariusheil, 2020-09-21:
Closing this as it was inactive for some time, feel free to reopen.

closed 2020-09-21 by mariusheil