bluerange-io/bluerange-mesh #146: Fruitymesh with i2c

mgprojetos, 2020-08-05:
Hi all.
How and where can we include i2c functions?

mariusheil, 2020-08-05:
Hi,

in your custom featureset header file, you should add:
```
//####### NRF Configuration #############
#define TWI_ENABLED 1
#define TWI1_ENABLED 1
```

In the featureset cmake file, you need to include the appropriate nordic drivers and your custom sources for a specific TWI/I2C driver (Check the paths:
```
target_sources(${FEATURE_SET} PRIVATE ${LOCAL_SRC})
target_include_directories(${FEATURE_SET} PRIVATE "${PROJECT_SOURCE_DIR}/src/c/_YOUR_DRIVER_PATH_TO_CUSTOM_FILES_")

if (${PLATFORM} STREQUAL "NRF52832")
  file(GLOB   LOCAL_SRC CONFIGURE_DEPENDS  "${PROJECT_SOURCE_DIR}/sdk/sdk14/components/drivers_nrf/twi_master/nrf_drv_twi.c"
                                           "${PROJECT_SOURCE_DIR}/sdk/sdk14/components/drivers_nrf/spi_master/nrf_drv_spi.cpp" 
                                           "${PROJECT_SOURCE_DIR}/sdk/sdk14/components/drivers_nrf/spi_master/nrf_drv_spi.c")
  target_sources(${FEATURE_SET} PRIVATE ${LOCAL_SRC})
elseif (${PLATFORM} STREQUAL "NRF52840")
  file(GLOB   LOCAL_SRC CONFIGURE_DEPENDS  "${PROJECT_SOURCE_DIR}/sdk/sdk15/modules/nrfx/drivers/src/nrfx_spi.c"
                                           "${PROJECT_SOURCE_DIR}/sdk/sdk15/modules/nrfx/drivers/src/prs/nrfx_prs.c"
                                           "${PROJECT_SOURCE_DIR}/sdk/sdk15/modules/drivers_nrf/twi_master/nrfx_twi.c"
                                           "${PROJECT_SOURCE_DIR}/sdk/sdk15/integration/nrfx/legacy/nrf_drv_spi.c"
                                           "${PROJECT_SOURCE_DIR}/sdk/sdk15/integration/nrfx/legacy/nrf_drv_twi.c")
  target_sources(${FEATURE_SET} PRIVATE ${LOCAL_SRC})
  target_include_directories(${FEATURE_SET} PRIVATE "${PROJECT_SOURCE_DIR}/sdk/sdk15/modules/nrfx/drivers/src")
  target_include_directories(${FEATURE_SET} PRIVATE "${PROJECT_SOURCE_DIR}/sdk/sdk15/integration/nrfx/legacy")
elseif (${PLATFORM} STREQUAL "ARM")
  # Arm has no implementation hence no additional files needed
else(${PLATFORM} STREQUAL "NRF52832")
  message(FATAL_ERROR "PLATFORM ${PLATFORM} does not support the asset module!")
endif(${PLATFORM} STREQUAL "NRF52832")
```

Then, in your configurationLoadedHandler, you should include the code to intialize the driver. This is the same way, that you are doing it when working with the standalone nordic SDK, so you can check on the devzone how that works.

Hope that helps,
Marius

mgprojetos, 2020-09-02:
Great!! It worked. Now I have I2C and SPI peripherals communicating with nRF52832. Thanks.
And what about UART? I need to connect my nRF52 with a microcontroller to change informations by UART and I don't need logs and flow control. How can I disable all the logs and the flow control?

mariusheil, 2020-09-02:
Hi,

in your custom featureset (or in the Github one if you are still using that), just define ACTIVATE_UART to 0. This will remove the uart code from the terminal. If you have ACTIVATE_SEGGER_RTT defined to 1, you will still be able to use the terminal over Segger RTT. You can then write your custom UART code.

Marius

mgprojetos, 2020-09-02:
I will try this. Thanks for the quick response.