bluerange-io/bluerange-mesh #147: Send custom messages to nodes

NilsMinor, 2020-09-04:
Hi,
I am building a library for communication between a central (my app) and some nodes (the mesh) --> [repo](https://github.com/NilsMinor/ngx-ble-mesh). I implemented the basic BLE stuff as well as the Encryption manager.  The next goal is to write a communication manager in order to send data to a node and receive data from it. Therefore I have some additional question.

- Is there an example how to construct a action-message? Like in the following example how will the packet structure look like? And am I right, when the packet is bigger then 16 bytes of data I must split it? What if the packet is more then 2 packets big, do I have multiple START packets?

```
       u8 data[1];
        data[0] = 123;

        //Send ping packet to that node
        SendModuleActionMessage(
                MessageType::MODULE_TRIGGER_ACTION,
                targetNodeId,
                PingModuleTriggerActionMessages::TRIGGER_PING,
                0,
                data,
                1,
                false
        );
```
- What is the best way to get all nodes from a mesh? I want to list up all nodes with their nodeIDs. Can I ask the node I connected how the mesh looks like?
- What is the right way to ask for data from a node? Using MODULE_TRIGGER_ACTION and MODULE_ACTION_RESPONSE ?

Many thanks,
Nils

Brotcrunsher, 2020-09-08:
Hi Nils! 

You did not explicitly mention it but from the first looks of your Encryption Manager implementation it looks like you are implementing the MeshAccessConnection counterpart for the App, which is certainly the correct approach. You probably already know about this documentation page, but in case you don’t, have a look at this: https://www.bluerange.io/docs/fruitymesh/MeshAccessConnection.html 
It explains the encryption logic in detail. 

Now to your explicit questions: 

- For action messages have a look at the connPacketModule type. It's an 8-byte header followed by the payload. MeshAccessConnections currently don't support MTU exchanges so yes, the size of the individual packets is limited to 16-byte (20 with encryption). Splitting works with the messageTypes SPLIT_WRITE_CMD and SPLIT_WRITE_CMD_END
- I don't quite understand what you mean by "... how the mesh looks like". Do you mean how the mesh is built up and which nodes are connected to which other nodes? If so, you can send out a broadcast of "action 0 status get_connections". Also have a look at the other terminal commands of the StatusReporterModule, they might give you the information that you are actually looking for.
- Yes, the intended way and the convention that FruityMesh follows (most of the time) is to send MODULE_TRIGGER_ACTION messages to a node which then answers with a MODULE_ACTION_RESPONSE. Have a look at the StatusReporterModule. get_connections follows exactly this pattern.

Hope that helped,
Jakob

NilsMinor, 2020-09-09:
Hi @Brotcrunsher ,
thanks for your support! 

Yes you are right, I already implemented the mesh access module. I am able to connect to the mesh using the encryption communication. What I am now doing is to implement the message schemes in order to read/write data from/to a specific node.

- Thanks, that helps my I will check the message splitting commands.
- yep you are right, I want to ask a node what other nodes a connected to the mesh in order to add them to my app and to be able to communicate with them.
- Ok thanks.

Your answers helped me thanks :)
BR,
Nils

closed 2020-09-14 by Brotcrunsher