bluerange-io/bluerange-mesh #148: Ho to get nodeId from broadcasting messages

NilsMinor, 2020-09-16:
Hi,

i am trying to get the nodeId from broadcasting messages. I receive two types of messages from the advertising packet, serviceData and manufacturingData. I implemented the serviceData that gives me information if the node is contactable or sink ... (https://www.bluerange.io/docs/fruitymesh/MeshAccessModule.html#_meshaccess_broadcast_messages).  And I receive the manufacturing Data ( 77 2 240 11 0 1 59 76 59 76 86 59 1 0 26 0 4 1 255 255 16 0 0 0 0 0). I think the [59 76] represents my nodeId I am looking for but I can not find the packet structure from the docs. Is this the correct way to get the nodeID?

Thanks and best regards,
Nils

mariusheil, 2020-09-17:
Hi, you should convert these numbers to hex format:
4D 02 F0 0B 00 01 3B 4C 3B 4C 56 3B 01 00 1A 00 04 01 FF FF 10 00 00 00 00 00

What your framework gave you is all the data of the manufacturer specific data field. The raw message would have contained 1 byte length field, then one byte type of the field (Manufacturer specific data 0xFF).

Now you have 0x024D which is the company identifier of Mway, then you have 0xF0 which is the mesh identifier, then you have 0x000B which is your network id, 0x01 is the messageType for the JOIN_ME packet, then 0x4C3B is your node id and so one....

What you got is the JOIN me message: https://www.bluerange.io/docs/fruitymesh/Specification.html#_join_me_packet_messagetype_1 with the header from here: https://www.bluerange.io/docs/fruitymesh/Specification.html#_advertising_packet_with_manufacturer_specific_data

Your framework might combine all the information of the two messages into one object, I am not sure. But what you are looking for is a message that contains the 16 bit Service UUID 0xFE12 (list of 16 bit service uuids) and you should get the service data. The MeshAccessBroadcast message does not use the manufacturer specific data field but rather uses the service data field.

See here for the different kind of fields:
https://www.bluetooth.com/de/specifications/assigned-numbers/generic-access-profile/

You should look for the following field instead of TYPE_MANUFACTURER_SPECIFIC_DATA = 0xFF:
TYPE_SERVICE_DATA                        = 0x16,

Marius

NilsMinor, 2020-09-17:
Hi Marius,

thanks that helped me, it is working ;)

BR,
Nils

closed 2020-09-17 by NilsMinor