bluerange-io/bluerange-mesh #149: Encryption Error: Wrong handshake packet

jjduhamel, 2020-09-24:
Hi,

I'm getting the following error after sending an snonce handshake packet in encrypted form.  

```
00> [MeshAccessConnection.cpp@1025 WARNING]: Wrong handshake packet
```

In plain form, the snonce packet should be 13 bytes long.  However, once I encrypt the packet, I get 16 bytes.  For clarity , I added some logging to verify that the data is decrypted properly.  However, although I'm receiving the correct data, the check at [this line](https://github.com/mwaylabs/fruitymesh/blob/master/src/mesh/MeshAccessConnection.cpp#L983) is causing the handshake to fail, because `sendData->dataLength` is 16 rather than 13.

*src/mesh/MeshAccessConnection.cpp*
```
        if(sendData->dataLength == SIZEOF_CONN_PACKET_ENCRYPT_CUSTOM_ANONCE && packetHeader->messageType == MessageType::ENCRYPT_CUSTOM_ANONCE){
            OnANonceReceived((connPacketEncryptCustomANonce const *) data);
        }
        else if(**sendData->dataLength == SIZEOF_CONN_PACKET_ENCRYPT_CUSTOM_SNONCE** && packetHeader->messageType == MessageType::ENCRYPT_CUSTOM_SNONCE){
            OnSNonceReceived((connPacketEncryptCustomSNonce const *) data);
        }
```

[Here](https://github.com/mwaylabs/fruitymesh/blob/master/src/mesh/MeshAccessConnection.cpp#L957), it looks like we simply subtract 4 from the payload size.  Shouldn't the AES128 algorithm produce a 16 byte (128-bit) blob when encrypting any packet up the the maximum 16 bytes?  How is sendData->dataLength supposed to be adjusted downward in the case where you're encrypting a 13 byte snonce packet?

**Fruitymesh Logs:**
```
00> [MeshAccessConnection.cpp@699 MACONN]: Decrypting 87:95:CD:B1:AD:99:56:8B:79:90:5E:53:64:85:21:8C:18:C5:76:7D (20) with nonce 541230638 00> [MeshAccessConnection.cpp@705 ERROR]:   - nonce: 541230638 
00> [MeshAccessConnection.cpp@708 ERROR]:   - nonce: 541230639 
00> [MeshAccessConnection.cpp@721 ERROR]:   - keystream1: FF:57:A9:C3:09:70:D1:7E:75:73:FE:03:13:A2:8A:3D 
00> [MeshAccessConnection.cpp@736 ERROR]:   - cleartext1: 78:C2:64:72:A4:E9:87:F5:0C:E3:A0:50:77:27:AB:B1 
00> [MeshAccessConnection.cpp@740 ERROR]:   - keystream2: 18:C5:76:7D:49:34:B8:94:17:3D:5E:E8:72:1D:34:6B 
00> [MeshAccessConnection.cpp@747 ERROR]:   - mic: 18:C5:76:7D 
00> [MeshAccessConnection.cpp@750 ERROR]:   - nonce: 541230638
 00> [MeshAccessConnection.cpp@764 ERROR]:   - keystream3: 9C:95:B0:F5:AD:99:56:8B:79:90:5E:53:64:85:21:8C
 00> [MeshAccessConnection.cpp@775 ERROR]:   - cleartext2: F1:12:37:3D:2E:86:42:20:00:00:00:00:00:00:00:00 
00> [MeshAccessConnection.cpp@779 ERROR]:   - nonce: 541230640 
00> [MeshAccessConnection.cpp@790 MACONN]: **decryptedOut: 1B:00:7D:44:00:00:00:00:00:00:00:00:00:00:00:00* *
00> [MeshAccessConnection.cpp@791 MACONN]: Decrypted as 87:95:CD:B1:AD:99:56:8B:79:90:5E:53:64:85:21:8C (16) micValid 1
```

mariusheil, 2020-09-25:
Hi,

did you already take a look at this documentation that explains the handshake in a lot of detail?
https://www.bluerange.io/docs/fruitymesh/MeshAccessConnection.html

In short, if the packet should only have 13 byte, you simply do not send the last three bytes. You are adding 0 padding to the cleartext packet and then xoring it with the keystream. Xoring will not scramble the bytes of the whole message and therefore you can simly cut off the last three bytes before sending the message.

Marius

jjduhamel, 2020-09-29:
Yep that worked.  Thanks for the quick response.  If it's all right to ask a question before you close this, I need to increment both the encryption and decryption nonces by 2 after I send or receive a packet.  Why increment by 2 rather than 1?

Also, this [spec](https://www.bluerange.io/docs/fruitymesh/MeshAccessConnection.html) says that the handshake packets are padded with zeros to be 16 bytes long, but the code linked above seems to require you not to send the trailing zeros.  Is this a typo in the docs or a planned change?

jjduhamel, 2020-09-30:
Ok I have it working now.  There was a bug in my code related to packet splitting, and my phone was disconnecting after like 30 seconds of the mic check failing.  Is this expected?  I'm wondering if there's a more graceful way to recover such as retrying with different nonces (+1, -1, etc.).

Here are the relevant logs:

```
00> [MeshAccessConnection.cpp@699 MACONN]: Decrypting 0A:08:3E:AA:2C:2D:3F:83:3C:3A:21:06:10:BA:5C:41:85:1F:B8:BD (20) with nonce 4197155770 00> [MeshAccessConnection.cpp@705 ERROR]:   - nonce: 4197155770 00> [MeshAccessConnection.cpp@708 ERROR]:   - nonce: 4197155771 00> [MeshAccessConnection.cpp@721 ERROR]:   - keystream1: 8C:81:C2:7C:2B:4C:1D:03:42:FA:82:AE:0A:4F:F5:2C 00> [MeshAccessConnection.cpp@736 ERROR]:   - cleartext1: 86:89:FC:D6:07:61:22:80:7E:C0:A3:A8:1A:F5:A9:6D 00> [MeshAccessConnection.cpp@740 ERROR]:   - keystream2: 85:1F:B8:BD:96:2D:68:71:93:FC:2D:E1:A3:5B:79:CC 00> [MeshAccessConnection.cpp@747 ERROR]:   - mic: 85:1F:B8:BD 00> [MeshAccessConnection.cpp@750 ERROR]:   - nonce: 4197155770 00> [MeshAccessConnection.cpp@764 ERROR]:   - keystream3: 1A:08:0D:AA:51:2D:3F:48:3C:3F:2D:06:10:BA:A3:41 00> [MeshAccessConnection.cpp@775 ERROR]:   - cleartext2: B8:A9:AB:E3:BA:83:2B:FA:00:00:00:00:00:00:00:00 00> [MeshAccessConnection.cpp@779 ERROR]:   - nonce: 4197155772 00> [MeshAccessConnection.cpp@790 MACONN]: Decrypted as 10:00:33:00:7D:00:00:CB:00:05:0C:00:00:00:FF:00:CD:EE:00:20 00> [MeshAccessConnection.cpp@791 MACONN]: Decrypted as 0A:08:3E:AA:2C:2D:3F:83:3C:3A:21:06:10:BA:5C:41 (16) micValid 1 00> [MeshAccessConnection.cpp@699 MACONN]: Decrypting 0A:09:0D:D5:AE:2D:3F:B7:C3:3F:2D:F9:6F:BA:A3:BE:77:9A:1C:79 (20) with nonce 4197155772 00> [MeshAccessConnection.cpp@705 ERROR]:   - nonce: 4197155772 00> [MeshAccessConnection.cpp@708 ERROR]:   - nonce: 4197155773 00> [MeshAccessConnection.cpp@721 ERROR]:   - keystream1: F6:35:91:03:45:4F:7D:23:BD:28:8B:C6:9F:A1:3C:E3 00> [MeshAccessConnection.cpp@736 ERROR]:   - cleartext1: FC:3C:9C:D6:EB:62:42:94:7E:17:A6:3F:F0:1B:9F:5D 00> [MeshAccessConnection.cpp@740 ERROR]:   - keystream2: 8B:1B:BE:66:DA:7A:3F:97:E1:18:D9:79:53:B1:53:A4 00> [MeshAccessConnection.cpp@747 ERROR]:   - mic: 77:9A:1C:79 00> [MeshAccessConnection.cpp@750 ERROR]:   - nonce: 4197155772 00> [MeshAccessConnection.cpp@764 ERROR]:   - keystream3: 88:E0:D1:32:59:45:66:36:BB:10:61:63:29:18:28:4B 00> [MeshAccessConnection.cpp@775 ERROR]:   - cleartext2: B8:A9:AB:E3:BC:83:2B:FA:00:00:00:00:00:00:00:00 00> [MeshAccessConnection.cpp@779 ERROR]:   - nonce: 4197155774 00> [MeshAccessConnection.cpp@790 MACONN]: Decrypted as 82:E9:DC:E7:F7:68:59:81:78:2F:4C:9A:46:A2:8B:F5:9D:EE:00:20 00> [MeshAccessConnection.cpp@791 MACONN]: Decrypted as 0A:09:0D:D5:AE:2D:3F:B7:C3:3F:2D:F9:6F:BA:A3:BE (16) micValid 0 00> [MeshAccessConnection.cpp@999 WARNING]: Invalid packet 00> [MeshAccessConnection.cpp@879 MACONN]: MA SendData from 68 to 4068 00> [MeshAccessModule.cpp@652 MACONN]: Message auth is 2 00> [MeshAccessConnection.cpp@125 MACONN]: Deleted MeshAccessConnection, discR: 0, appDiscR: 1 00> [MeshAccessModule.cpp@344 MAMOD]: Broadcasting mesh access 02:01:06:03:03:12:FE:0F:16:12:FE:03:00:0B:00:09:35:F7:28:00:00:00:00, len 23 00> {"type":"update_joinme","clusterId":2945646660,"clusterSize":1} 00> [GAPController.cpp@119 C]: Disconnected device 22 00> {"type":"live_report","nodeId":68,"module":3,"code":52,"extra":0,"extra2":22}
```

closed 2020-09-30 by jjduhamel