bluerange-io/bluerange-mesh #150: How to write the network key

jjduhamel, 2020-09-30:
Hi, I no have encryption working.  The final step I'm trying to figure out is how to set the network key.  I made the following change to the code; however, I added some logging to the firmware and it seems to be using 0xefef...efef.  What's the correct procedure to program a device with a randomly generated network key?

```
$ git diff config/featuresets/github_nrf52.cpp
diff --git a/config/featuresets/github_nrf52.cpp b/config/featuresets/github_nrf52.cpp
index 591c455..f09e0dd 100644
--- a/config/featuresets/github_nrf52.cpp
+++ b/config/featuresets/github_nrf52.cpp
@@ -66,7 +66,7 @@ void setFeaturesetConfiguration_github_nrf52(ModuleConfiguration* config, void*
         c->enrollmentState = EnrollmentState::ENROLLED;
         c->networkId = 11;
         // XXX: Randomly generated network key.  Change this if it's compramised.
-        CheckedMemset(c->networkKey, 0x00, 16);
+        CheckedMemset(c->networkKey, 0xdeadbeefdeadbeefdeadbeefdeadbeef, 16);
     }
 }
```

mariusheil, 2020-09-30:
Hi,

the correct procedure if you have a lot of nodes that go into production is to write the necessary data in the UICR, see here: https://www.bluerange.io/docs/fruitymesh/Specification.html#UICR

The idea in general is to have a unique (randomly generated) node key (aka. device key). Then the user will use an enrollment application or gateway to enroll all nodes into one network (See Enrollment Module documentation).

If you just want to flash some nodes and have them in the same network, you can also write a networkId > 1 in the UICR and a network key. Then all these nodes wil automatically be in the same network after flashing.

Marius

jjduhamel, 2020-10-02:
Ok I see.  It would be helpful if you provided an example of using `srec_cat` to write the nodeKey, networkKey, etc.  Is it safe to assume that all my devices are currently programmed with the aforementioned network key (0xef...ef)?

Also, what's the purpose of the following line?

`CheckedMemset(c->networkKey, 0x00, 16);`

mariusheil, 2020-10-02:
Hello,

that is a good idea. I will try to get an example together and write it here. The CheckedMemset is currently setting the networkkey to 000000...000000. I will try to set this straight and document it a bit better.
I'll respond again once I have that.

Marius

jjduhamel, 2020-10-02:
Thanks.  Look forward to your response.

mariusheil, 2020-10-02:
Hi, I have updated our documentation with a lot more clarification and an example on how to use srec_cat to create the UICR. I have attached this information to the ticket and it will be available officialy soon after we release again.

[Quick Start.pdf](https://github.com/mwaylabs/fruitymesh/files/5317287/Quick.Start.pdf)
[Specification.pdf](https://github.com/mwaylabs/fruitymesh/files/5317288/Specification.pdf)
[Developers.pdf](https://github.com/mwaylabs/fruitymesh/files/5317289/Developers.pdf)

Marius

mariusheil, 2020-12-08:
Hi, I will close this as it was inactive for a while.

closed 2020-12-08 by mariusheil