bluerange-io/bluerange-mesh #151: System is rebooting in loop

NilsMinor, 2020-09-30:
Hi,

I updated my fruitymesh code to the latest version/master and now my board is rebooting in a loop. I followed the new cmake instructions to build the software and it works great in vscode!

After flashing I get this output :

```
—————————————————————————
{„type“:“reboot“,“reason“:0,“code1“:0,“stack“:0,“version“:84250,“blversion“:0}
[FruityMesh.cpp@245 MAIN]: Reboot reason was 0, SafeBootEnabled:0, stacktrace 0
[FruityMesh.cpp@249 MAIN]: Current Stack position: 536936332, Stack Limit: 536924824, Stack max size: 11508
[FruityHalNrf.cpp@283 NODE]: Init Softdevice version 0x3, Boardid 10000
[2J[H—————————————————————————
```
My code /changes are here [https://github.com/NilsMinor/fruitymesh/tree/firelight](https://github.com/NilsMinor/fruitymesh/tree/firelight)

I do not really get what has changed and why the system is always rebooting, maybe some one can help me.

Thanks,
Nils

mariusheil, 2020-09-30:
That are the only lines you are getting? Then it looks like something in the softdevice initialization goes wrong. Do you have debugging working? if yes, set a breakpoint in the hardfault and app fault handlers. Try to activate the DEBUG define and make sure that the log in the hardfault handler is activated. remove the system reset from the fault handlers and put an endless loop in there to check why it is rebooting.
I do not know what might be causing the error as it says reboot reason is 0, which is unknown, mostly because of power loss.

Typical stuff that could go wrong during softdevice initialization is that a dcdc was configured for the board but there is none or that a low frequency external crystal was configured but there is none on the board.

mariusheil, 2020-09-30:
Maybe try the SYNTH clock source and see if it boots. Maybe also add a few logs to log out your board settings before the softdevice is initialized. Maybe the board settings are not properly loaded.

NilsMinor, 2020-09-30:
Hi Marius,

i am not able to debug yet but maybe I will get it running. Your advice with the dcdc is good, my board has no dcdc connection that's why i disabled it in the board config (https://github.com/NilsMinor/fruitymesh/blob/58666df6637dc9a6d3397315f6d9f5c2de61ad12/config/boards/board_inuk.cpp#L44) Maybe this is setting is not used anymore or is there an other way to disable the dcdc?

Yes this is the only output it really seems to be the dcdc converter.

mariusheil, 2020-09-30:
You should probably check the lines where the softdevice is activated and log out all the parameters that are passed to it. Something is probably being passed in the wrong way, then log the error code returned from the softdevice before APP_ERROR_CHECK is executed because this will call the fault handler that will then reboot the system.

NilsMinor, 2020-10-01:
Hi Marius,

i found my bug, I added custom pins in the old way so I followed the new CustomPins scheme and it is working now :) thanks again for your help!

closed 2020-10-01 by NilsMinor