bluerange-io/bluerange-mesh #154: Why is the use of the heap prohibited?

nishinohi, 2020-10-16:
Is the reason for [prohibiting the use of the heap](https://www.bluerange.io/docs/fruitymesh/Specification.html#_heap_usage) to prevent memory leaks by contributors or others?
Or is it an architectural constraint on Fruitymesh?

mariusheil, 2020-10-16:
Hi,

we just prevented it as we think it leads to better design choices and less memory consumption. Debugging heap issues can be quite complicated and a fragmented heap could be a waste of space.
You are free to use the heap if you need to and you can circumvent this restriction by adding this to your featureset's CMake file:

`set(ALLOW_MALLOC 1)`

You should only do this if you know what you are doing.

Marius

nishinohi, 2020-10-16:
Thank you for answering.

> Debugging heap issues can be quite complicated and a fragmented heap could be a waste of space.

Convinced as an answer to my question.

---

This is not a question directly related to the Fruitymesh project, so I would be grateful if you could give me your opinion on the following:

`What implementation would you prefer if you want to work with variable length arrays?`

The background of this question is as follows.

I am creating a custom module to operate a cellular module with AT commands via UART. (Creating a minimal gateway)
To perform a specific function (for example, opening a socket), I need to execute a series of AT commands and check the success of each command. Since each AT command is a very small function, I think that implementing the method individually should be avoided because it makes the custom module unnecessarily bloated. Therefore, I thought about saving a series of commands in a variable length array and executing the commands while monitoring the UART response if the array is not empty.

mariusheil, 2020-10-16:
Hi,

we have a macro in our code called DYNAMIC_ARRAY which reserves a variable length array on the stack. We use that quite often as it is a nice way where we do not need to keep track of the array as the memory is freed after the function is left.

I am not very familiar with AT commands. Ist it sth. synchronous where you will receive an answer to each request? If yes, I would just choose a buffer that is big enough to hold the largest command and put request/responses in there.

We have a number of implementations for customers that impement UART protocols. In most of these implementations, we need to Queue a number of variable length commands and want them to be executed one after the other. We use a Wrapper around the PacketQueue class for that. A method that queues the data into the write queue and another method that processes the incoming data and that checks the last element from that queue and matches requests and responses. The packet queue simply allocates a buffer that can fit a number of packets.

Marius

nishinohi, 2020-10-16:
DYNAMIC_ARRAY may not be appropriate for use in this case as it is necessary to keep the command stack outside the scope of the function. However, DYNAMIC_ARRAY is very convenient and very helpful!

> synchronous where you will receive an answer to each request? 

yes.

I haven't figured out the details of the PacketQueue class yet, but after reading your description, I feel that creating a wrapper could fulfill my request.

Thank you for your detailed answer

closed 2020-10-16 by nishinohi