bluerange-io/bluerange-mesh #155: Calculating the node distance based on rssi value

NilsMinor, 2020-11-01:
Hi,

I found an approach to calculate the distance between two ble devices using the rssi (https://iotandelectronics.wordpress.com/2016/10/07/how-to-calculate-distance-from-the-rssi-value-of-the-ble-beacon/) have you done something similar using fruitymesh/bluerange? Do you have an other approach or an advice how to do this for example how to get the rssi value of an other node?

Thanks,
Nils

mariusheil, 2020-11-01:
Hi,

yes. We have a full asset tracking solution, see https://BlueRange.io

For a simple approach, you can use e.g. "action 0 status get_connections"
to get the RSSI of all connections between nodes. Use "action 0 status
get_nearby" to get a a list of RSSI values for all nodes in the
surroundings that are in the same network.
Take a look at the scanning module on how to generally scan for any kind of
packet and get the RSSI and then send this through the mesh.

Be aware that the RSSI is quite inaccurate when transformed to a distance.
Do not expect wonders. We are doing quite  a lot of magic on our side in
the commercial solution to get good estimates. For a simple version expect
to get a distance that is accurate by +/- 5 to 10 meters. You can mostly
estimate if sth. is very close within a centimeter, a meter or 10 meters.
If you estimate bigger distances , they are mostly inaccurate.

Marius

NilsMinor, 2020-11-02:
Hi @mariusheil ,

thanks that's constructive feedback! My nodes are maybe 5 to 15 meters a way. I think I will give it a try or I maybe will use an other way to get/set the distance between my nodes.

Thanks,
Nils

closed 2020-11-02 by NilsMinor