bluerange-io/bluerange-mesh #156: Bluetooth 5 Long Range

marcantoinebouchard, 2020-11-27:
Hi,

The problem I have trying to make it work with Long Range is there is no "Connectable and scannable undirected" advertising type in Bluetooth 5 Long Range spec. Does the FruityMesh design lie on this advertising type?

Thanks,
Marc

mariusheil, 2020-12-03:
Hi,

no, the design of FruityMesh does not rely on this type of advertising. You need to have the possibility to distribute an advertising packet that informs surrounding node about the clusterSize and some other information. Also, you need to be connectable (of course). If these conditions are met, you can probably implement long range support.

It is on our roadmap as well, but we have a lot of other requested features that we are working on and it hasn't yet gotten a high enough priority internally.

Marius

mariusheil, 2020-12-23:
Closing this for now. If we have more to share such as a roadmap, we will announce it.

closed 2020-12-23 by mariusheil