bluerange-io/bluerange-mesh #158: Where are module IDs defined?

Rewbert, 2020-12-07:
Hi!

Where are module ID's for the different modules defined? E.g 1 for Beaconing module. I can see that a module id is passed in as an argument to the constructor but I can't trace it and find a call site.

Trying to wrap my head around how modules work so that I can write an example module of my own.

Thanks

Rewbert, 2020-12-07:
Nevermind, I found it the second after I wrote this issue. It is in PrimitiveTypes.h.

closed 2020-12-07 by Rewbert

mariusheil, 2020-12-07:
Hi,

make sure to read the documentation on module ids:
https://www.bluerange.io/docs/fruitymesh/Specification.html#ModuleIds

Also, you can follow this tutorial:
https://www.bluerange.io/docs/fruitymesh/Implementing-a-Custom-Module.html