bluerange-io/bluerange-mesh #160: Probable error in tutorial

Rewbert, 2020-12-08:
Hi,

When following [https://www.bluerange.io/docs/fruitymesh/Implementing-a-Custom-Module.html](this) tutorial the process comes to a halt at stage three. CMake will build the `PingModule.cpp` found in `src/examples/` as well and then complain about multiple definitions found in that file and the one the tutorial just instructed you to create.

Robert

mariusheil, 2020-12-08:
Hi,

you are right. This is because we decided to include the PingModule in our build process to make sure that it will always compile. We have not thought of that. You should be able to just delete The PingModule.h and PingModule.cpp from examples folder and it should work. We will add that step to the tutorial if it works.

Marius

Rewbert, 2020-12-08:
Thanks, that's what I did and it worked. There's another issue in step 5 I believe. It tells you to write
```
void PingModule::MeshMessageReceivedHandler(BaseConnection* connection, BaseConnectionSendData* sendData, connPacketHeader* packetHeader)
{
    //Must call superclass for handling
    Module::MeshMessageReceivedHandler(connection, sendData, packetHeader);

    //Filter trigger_action messages
    if(packetHeader->messageType == MessageType::MODULE_TRIGGER_ACTION && sendData->dataLength >= SIZEOF_CONN_PACKET_MODULE_VENDOR){
        connPacketModuleVendor* packet = (connPacketModuleVendor*)packetHeader;

        //Check if our module is meant and we should trigger an action
        if(packet->moduleId == vendorModuleId){
            //It's a ping message
            if(packet->actionType == PingModuleTriggerActionMessages::TRIGGER_PING){

                //Inform the user
                logt("PINGMOD", "Ping request received with data: %d", packet->data[0]);

                //TODO: Send ping response
            }
        }
    }
}
```
But the lowercase c on `connPacketModuleVendor` should be uppercase.
```
void PingModule::MeshMessageReceivedHandler(BaseConnection* connection, BaseConnectionSendData* sendData, connPacketHeader* packetHeader)
{
    //Must call superclass for handling
    Module::MeshMessageReceivedHandler(connection, sendData, packetHeader);

    //Filter trigger_action messages
    if(packetHeader->messageType == MessageType::MODULE_TRIGGER_ACTION && sendData->dataLength >= SIZEOF_CONN_PACKET_MODULE_VENDOR){
        ConnPacketModuleVendor* packet = (ConnPacketModuleVendor*)packetHeader;

        //Check if our module is meant and we should trigger an action
        if(packet->moduleId == vendorModuleId){
            //It's a ping message
            if(packet->actionType == PingModuleTriggerActionMessages::TRIGGER_PING){

                //Inform the user
                logt("PINGMOD", "Ping request received with data: %d", packet->data[0]);

                //TODO: Send ping response
            }
        }
    }
}
```

Rewbert, 2020-12-09:
Another error - it appears that we are fetching the target node to ping from the wrong index.
Step four tells us to paste in this code:
```
if(TERMARGS(0, "pingmod")){
    //Get the id of the target node
    NodeId targetNodeId = Utility::StringToU16(commandArgs[0]);
    logt("PINGMOD", "Trying to ping node %u", targetNodeId);

    //TODO: Send ping packet to that node

    return TerminalCommandHandler::SUCCESS;
}
```
But `targetNodeId` becomes `0` every time (for me), which will broadcast the `ping` request to every node in my network. I believe this element would contain the word `pingmod` rather than the node id - is this correct?

Replacing the `0` with a `1` solves the issue for me. Only the node whose Id I write will get the ping request.
```
if(TERMARGS(0, "pingmod")){
    //Get the id of the target node
    NodeId targetNodeId = Utility::StringToU16(commandArgs[1]);
    logt("PINGMOD", "Trying to ping node %u", targetNodeId);

    //TODO: Send ping packet to that node

    return TerminalCommandHandler::SUCCESS;
}
```

mariusheil, 2020-12-09:
Hi,

thanks for that one as well. I have just looked that up in the PingModule.cpp where it is correct but it looks like this is wrong in the tutorial. We will fix this.

Marius

mariusheil, 2020-12-23:
Hi, documentation is fixed now. Feel free to reopen it, if there are other issues :-)

closed 2020-12-23 by mariusheil