bluerange-io/bluerange-mesh #162: 8dB defaultDBmTX

maagedmazyek, 2020-12-15:
Is it possible to make use of the 8dB (NRF52840) and have a more stable connection?

Anyway I tried to set `defaultDBmTX = 8;` in Config.h (latest firmware), and it results in a reboot-loop of the firmware.

mariusheil, 2020-12-15:
Hi,

I would need to look into this with more detail, but it is probably rebooting because the SoftDevice does not accept the given value. The error probably happens in FruityHalNrf::BleStackInit where the method RadioSetTxPower is called. It probably returns an error code, so you should check if that value was correctly translated by our HAL or not and what the returned error is. Just add a log statement there and you will find out.

Actually, from looking at the implementation of RadioSetTxPower, I can see that the value of 8 is not properly mapped by our HAL to the nRF52840. We will fix that, I created a ticket. In the meantime, you should be able to add the value yourself.

Marius

maagedmazyek, 2020-12-15:
Ah, this looks simple. I was not sure if RadioSetTxPower is the only spot. I will report.

maagedmazyek, 2020-12-15:
It works. First impression: more stability / greater range, but this has to be tested more.

Next thing I'm going to do is to measure the energy consumption difference. Im expecting 20-40% increase here.

mariusheil, 2020-12-23:
Hi, I am closing this now, feel free to reopen it if something pops up again.

closed 2020-12-23 by mariusheil