bluerange-io/bluerange-mesh #163: Discovery state does not change to LOW

nishinohi, 2020-12-25:
Assumption: master v0.8.4830

I set `highToLowDiscoveryTimeSec` to 30 sec and tried to see if the Discovery state changed from HIGH to LOW. However, it does not change to LOW because `KeepHighDiscoveryActive()` is called at intervals shorter than 30 seconds. `KeepHighDiscoveryActive()` is called from the Node module's `FindTargetBuffer()`, but I'm not sure why it's called so often because I'm not sure what this method does.

I would like to know the role of `FindTargetBuffer()`, meaning of `space` in the comment of `FindTargetBuffer()` and what kind of situation it is when `targetBuffer->payload.sender` is 0.

nishinohi, 2021-01-04:
I have confirmed that this issue does not occur in `master v1.0.90`.
I don't understand the behavior of `FindTargetBuffer ()`, but if it is not a serious problem, you could close this issue.
Thank you.

mariusheil, 2021-01-06:
Hello,

glad it was solved. I am currently on vacation. I will leave the issue open and will follow up with some information on the method once I am working again.

Marius

mariusheil, 2021-01-28:
Hi, the FindTargetBuffer method is meant to put all potential partners for a connection into a list. The clustering algorithm will then select the best partner from this list every once in a while. "Space" simply means an entry in this list. It is first looking if an entry for that node already exists and can be updated, then for an empty slot to fit in the data, etc,....

sender == 0 was simply used to check if the slot in the list is empty or not so that it does not need to do a full memory compare.

I'll close the ticket but feel free to reopen it if sth. is unclear

closed 2021-01-28 by mariusheil