bluerange-io/bluerange-mesh #164: nrf52840 dongle

giowild, 2021-01-14:
Hi,
thanks for the great software!

I have a question:
Is any of the 4 precompiled firmware provided in the release, known to work with the nrf52840 dongle?

I tried to flash the "nrf52-sink" on the nrf52840 dongle (using DK and Tagconnect TC2050 connector), but it seems that the dongle is now stuck in 1.8v mode and it doesn't allow any more reading or flashing of data through the nrf connect programmer app.

Is this something that you would expect by using the precompiled binaries provided and their UICR?

What is the fastest way and type of connection to flash fruitymesh repeatedly to the nrf52840 dongle?

By reading the following tutorial from Nordic, it seems that once in 1.8V I cannot recover it anymore, unless I cut and solder two pins. 
https://devzone.nordicsemi.com/nordic/short-range-guides/b/getting-started/posts/nrf52840-dongle-programming-tutorial

Can you confirm this is the only way?

Thanks in advance!

mariusheil, 2021-01-14:
Hi,

sorry, no. None of the provided firmwares was meant for the nrf52840
Dongle. The nrf52840 Dongle is sadly a bit of a tricky thing to use as it
does not have an on board debugger. Once a firmware was flashed that was
compiled for another board, you will have to follow the documented recovery
procedure that you have found.
It is certainly possible to create a fruitymesh version that is compatible
with the dongle. I will propose internally that we crate a precompiled
version for that board.
Once you resolder the board, you can repeatedly flash the dongle with the
firmware but you have to compile it yourself. If you include the correct
settings for the dongle, you can also use the 3.3v mode. You will have to
follow the Nordic guidelines for now.
For using UART over USB, you will need a special USB-CLC implementation as
well. We have it ready but it might not be included in the current release
yet, we will probably release it soon then.

Marius

giowild, 2021-01-15:
Thank you for answers. 
I will follow the cut & soldering procedure then.

What I'm trying to have is a rapid development setup, withouth the need for attaching & detaching cables between flashing & testing. The dongle should be able to be connected from one side, via usb, to a rapsberry pi and to establish serial communication with it. At the same time, dongle should be connected via Tagconnect cable to a DK and able to be flashed.

Do you think this is something that can be accomplished or it is simply impossible to have both connection in place for power supply reasons or other hardware related reasons?

It would not be an issue to compile fruitymesh by myself. I already have vscode and other files in place and firmware seems to compile correctly!! You did a very nice job!! 

Do you have settings for the dongle to share or USB CLC implementation? 
Are there some specific function already included in fruitymesh to enable 3.3V mode? 
At which point of the code would you suggest to call these 3.3V enabling functions?

Thanks a lot!

mariusheil, 2021-01-15:
Hi,

the best development setup for your case will be an nrf52840-DK. It has 2
micro usb ports, one for debugging and another for the "Dongle" connection.
I used that for developing the usb CLC communication and it works very
well. You can use the Dongle as well and I think it is possible to connect
both the debugger and the usb connector of the dongle at the same time. I
have not tried this, as we are using a different dongle (from Laird) but it
is almost the same hardware.
There are currently no 3.3v dongle settings in fruitymesh, but that should
only be 3 lines of code or so. Just call the lines of code in the
initialization of your custom module.
I can probably share the CLC implementation but I will have to discuss this
internally next week. I will report back.

Marius

giowild, 2021-01-17:
Hi, 

Great! Thanks to your help I managed to recover my dongles from 1.8V to 3V and then flash them again with bootloader. Now I can flash via USB DFU but also via TagConnect cable and have the dongle attached at the same time to test serial communication. Now what I need is to implement USB serial communication with raspberry. If you can discuss about it internally would be great!

Thanks a lot!
Giovanni

giowild, 2021-01-26:
Hello,
do you have some news on opening CLC implementation?

Thanks,
Giovanni

mariusheil, 2021-01-26:
Hello,

we have planned to open source the implementation and it will probably be planned in for the coming two weeks. It will most likely be in the next release before the end of february.

Marius

giowild, 2021-01-27:
Ok great! 
I'll wait for it!

Thank you very much! 
Giovanni

mariusheil, 2021-03-09:
Hi, we didn't make it to the end of february, but it is part of the latest release from 3rd march. I'll close this ticket. Feel free to open another one if you experiance issues.

Marius

closed 2021-03-09 by mariusheil