bluerange-io/bluerange-mesh #166: Handshake

mgprojetos, 2021-01-27:
Hi.
I have a 100 nodes network with several custom modules, and everything is workng fine.
My question is about handshake. I noted a lot of handshake in 10 minutes intervals. Is ti normal? Can I change this interval?
Thanks
Mário

mariusheil, 2021-01-27:
Hello,

can you provide me with more information on where you noticed the handshakes. E.g. a log or sth. similar. What is your setup, are the nodes close to each other, widely distributed. Some of them far away?

FruityMesh will do connection reestablishment to heal broken routes and if everything is working fine for you, you do not have to worry about that. Most of the time, packets will be queued and you will only experience an increased latency for some time. We will be improving on the stability of the connections more and more as well.

Marius

mgprojetos, 2021-01-28:
I've been thinking in a possible cause:
I have 20 nodes very closed, side-by-side, about 10 meters from the gateway, all static, no movement. I never positioned the nodes away one from another, and aways tested in group very closed one from another. Maybe an adjust from each node caused by the proximity can generate a high number of interactions and handshaking?
Attached a file with the serial messge during handshakes period
[log_handshake_1.txt](https://github.com/mwaylabs/fruitymesh/files/5887538/log_handshake_1.txt)
[log_handshake_2.txt](https://github.com/mwaylabs/fruitymesh/files/5887539/log_handshake_2.txt)

mariusheil, 2021-01-28:
As there are no time stamps in the log, what is the time frame of one of these logs? Looks like there are quite a few instabilities. What I can see very often in the log is "Disconnected 0 from connId:2, HCI:19 Remote User Terminated Connection" which means that the partner node has disconnected the connection on purpose so you would need a log of the partner node to check why it did that.
In log 2 I can see the line "CONN 0 disconnected because it had the same clusterId before handshake" which is a very weird error to happen frequently. That can only happen if the node has not updated its advertisement message in time and another node connected to it on wrong assumptions.

What you should try is:
- Make sure you use the latest version of FruityMesh
- Temporarily disable all your custom modules and see if it still behaves instable (best to completely remove them from the build to make sure everything is 100% removed) or just use the checked our version.
- If it still behaves instable, it could be because the hardware you are using has e.g. a different block crystal and the accuracy is wrong or some other issues.

Marius

mgprojetos, 2021-01-28:
Thank you so much for the quick answer.
The time frame of one of these logs is about 1-2 minutes.
Ok, I will check this.
Thanks again.

mariusheil, 2021-06-23:
Hello, I'll close this as there has not been any more feedback. Feel free to reopen it.

closed 2021-06-23 by mariusheil