bluerange-io/bluerange-mesh #167: Android App for FruityMesh

nishinohi, 2021-02-01:
I'm working on an Android application that connects nodes and smartphones via MeshAccessConnection (encrypted handshake is already supported). It is working for the most part, but can I make it open source?
I'm not sure if I should refrain from open sourcing it, for example, if you plan to release a paid smartphone application for FruityMesh.

mariusheil, 2021-02-01:
Hello,

if you could open source it, that would be awesome 🙂 We have been planning
to open source an Android SDK for ages but have not yet been able to do so.
We believe in open source as the best way to extend the ecosystem and are
always trying to share as much as possible from our codebase 🙂

Marius

nishinohi, 2021-02-01:
I will publish it as OSS.
However, the purpose and function of my smartphone app is only to connect to any node via MeshAccessConnection and send/receive some data. (The main purpose is to run the EnrollmentModule through the smartphone app.) I don't plan to port all the functions of FruityMesh to Android, so it may not be what you are expecting.

Thank you.

closed 2021-02-01 by nishinohi

mariusheil, 2021-02-01:
Hi, that is fine. I am sure it will help others if they try to implement something similar. Thanks