bluerange-io/bluerange-mesh #169: Connecting to serial terminal from Linux/picocom

Michaelhobo, 2021-02-07:
I'm trying to connect to my nrf52840 with fruitymesh's precompiled binary flashed to the device.

I pulled down and flashed github_dev_nrf52840.hex, and I'm connecting using picocom.

```bash
sudo picocom /dev/serial/... -b 1000000 -f h
```

It responds to <Enter>, but I can't get it to show status or do anything else. I read in another bug that flow control needs to be set to RTS/CTS, which I did. I also read that commands need to send \r, which I think I've done...

Here's my terminal log:

```
flowcontrol    : RTS/CTS
baudrate is    : 1000000
parity is      : none
databits are   : 8
stopbits are   : 1
escape is      : C-a
local echo is  : no
noinit is      : no
noreset is     : no
hangup is      : no
nolock is      : no
send_cmd is    : sz -vv
receive_cmd is : rz -vv -E
imap is        : 
omap is        : 
emap is        : crcrlf,delbs,
logfile is     : none
initstring     : none
exit_after is  : not set
exit is        : no

Type [C-a] [C-h] to see available commands
Terminal ready

mhTerm: status
mhTerm: status\r
mhTerm: status\n
mhTerm: status \r\r\n
mhTerm: status\r\n
mhTerm: status
mhTerm: 
mhTerm: 
mhTerm: 
mhTerm: 
mhTerm: 
mhTerm: 
mhTerm: 
mhTerm: 
mhTerm:
```

Does anybody have pointers for what/how to debug this problem?

mariusheil, 2021-02-07:
Hi,

I don't have a good tip for that program, but if you have access to a different serial terminal you could try that. Maybe you have access to a Windows computer and could try with putty as this is known to work nicely. As the terminal seems to be outputting correctly, I would guess something with the transmitted line endings is still wrong. You could of course attach a logic analyzer or try to log the sent characters in raw format but that might be a bit of work.

Marius

Michaelhobo, 2021-02-07:
Thanks for the pointers. It works with Windows+Putty. I'll experiment with different line endings, to see if one of picocom's options works.

> access to a different serial terminal

Do you know of a terminal from linux that's worked for others? I usually use picocom out of habit, but I'm happy to try another, as long as it's Linux based. I don't think my peers will be very happy with using Windows.

mariusheil, 2021-02-08:
Hi,

there are tons of terminals for sure that will work, I was mostly using minicom on linux so far. it is not the most user friendly terminal however.

Marius

Michaelhobo, 2021-02-08:
minicom worked out of the box, you've now converted me to minicom. Thanks for the pointers!

Michaelhobo, 2021-02-24:
I'm not sure what happened, but minicom is no longer working. When I switch back to Putty on Windows, it works just fine. I also tried putty on Linux, and it still didn't work.
I'll try to dig around more and I'll put my findings here afterwards.

mariusheil, 2021-06-23:
Hello, I'll close this as there has not been any more feedback. Feel free to reopen it.

closed 2021-06-23 by mariusheil