bluerange-io/bluerange-mesh #170: Broadcast Enrollment

nishinohi, 2021-02-13:
The EnrollModule needs to send a message to each node specified by serialNumber among multiple nodes. Therefore, I understand that if I don't enroll in the flash of the program, I need to send a message to the number of nodes to enroll all nodes.

What I want to do is to enroll the nodes that are interconnected in the mesh with a single broadcast message. 

The following scenario assumes that it is unlikely that anyone else will connect to the node (for security reasons).
The node has a default network key and network ID, and automatically creates a mesh when it starts. It broadcasts a message containing the `node ID offset value` (see below), the network ID, the `current cluster size`, and the required security key(network key, etc.). The node that receives the message saves the contents and broadcasts a response containing its own serialNumber. The node that receives the response compares its own serialNumber with the received one, and if its own is larger, it increments the `node ID offset value`. After receiving the same number of responses as the `current cluster size`, it uses the `node ID offset value` as the new node ID and enrolls itself with the saved content.

The node IDs in the mesh are not duplicated in this way if the assumption is that the serial numbers are not duplicated. The `node ID offset value` is used as an initial offset value when, for example, there are 5 nodes that have already been enrolled and you want to start the node ID of the newly enrolled node at 6.

Are there any security issues, or any other serious problems with this method?

The implementation is already complete and is working as expected. If this feature is useful, I'd like to pull request it.

mariusheil, 2021-02-15:
Hello,

I am not sure if I correctly understand what you proposed. I have written down a few thoughts of mine.

The idea of the enrollment module is to be able to add individual devices to a network. For our production devices, the use usually uses some form of QR-Code that he scans for each device and then adds it to the network. All of our production devices are unenrolled (aka. unprovisioned) once the customer buys them. He can therefore buy any number of devices and can securely add these to his secured network.

For testing purpose and for small projects, the GitHub featureset has an enrollment hardcoded that allows all flashed nodes to connected to each right after they were flashed.

A few years ago we had the possibility to enroll all devices that are connected to a so called"provisioning network" into a network of their own. This has a few drawbacks however:
- You never know who is currently connected to this provisioning network and who is not
- Anyone can spy on your enrollment message as he might know the key of your provisioning network (It will probably be public after some time if the same key is flashed on all devices)
- If you have the devices scattered through a building and you enroll the first devices, you might loose access to the other devices somewhere else in the building as these are now in a different network.

This might however be a suitable and easy method for some projects and I am sure there are use-cases for it. We have decided to not use the approach anymore, but if the problems that I mentioned do not apply for your project, then go with it :-) However, I think that we will probably not include this in our code as it is hard to explain the differences and we have had our process working and tested for quite a while and know that it works well for most use-cases.

Thanks,
Marius

nishinohi, 2021-02-21:
> A few years ago we had the possibility to enroll all devices that are connected to a so called"provisioning network" into a network of their own. This has a few drawbacks however:

The function I have implemented is probably similar to the above.

I think that the problems you described are probably not solved by my implementation either. So I won't send a pull request or anything like that. Thank you for your valuable input.

As this is not the purpose of this issue, I would like to know if you can tell me how to register using a QR code, which has random key information printed on it, read it with a smartphone app, etc., and then use that key to connect and register via MeshAccessConnection, etc. Does that mean I can connect and register? I thought that would certainly be a safe and hassle-free method.

Thanks

mariusheil, 2021-02-22:
Hello,

I have attached our documentation for QR code Generation. This will also be part of the official documentation in our next release. I order to connect to the Node, you will have to use the MeshAccess connections. Make sure to read our documentation on the UICR for more information on the node key: https://www.bluerange.io/docs/fruitymesh/Developers.html#UICR and https://www.bluerange.io/docs/fruitymesh/Specification.html#UICR

[QR Code Generation.pdf](https://github.com/mwaylabs/fruitymesh/files/6020819/QR.Code.Generation.pdf)

Marius

closed 2021-03-12 by nishinohi