bluerange-io/bluerange-mesh #171: Creating simple simulation with a cherrysim tester

mabner1996, 2021-02-26:
Hello, I would like to run a simple simulation with a cherrysim_tester.
There are several things I want to ask:

1. Is there any documentation of how to make simulation in the cherrysim_tester. Whenever I run the cherrysim_tester.exe, it will go through all the example in the "fruitymesh-master/cherrysim/test/" directory

2. So what I would to simulate is :
       a. Load a json file configuration I already prepared
       b. run the simulations for a timeline, for example 5 minutes
       c. input some terminal command in the timeline, for example I want node 1 to send data to node 2 during minute 1.

Any tips on how to create the script for the simuation? Thanks

mabner1996, 2021-03-24:
Hello I would ask if there is a way to make a simple simulation in the cherry sim using the tester, thank

mariusheil, 2021-03-24:
Hello,

sorry for not answering sooner. I think your initial question went under the radar.
Take a look at the test files in `fruitymesh-master/cherrysim/test/` and especially at the `TestClustering.cpp` file. We don't have a guide on how to write the tests, but there are plenty of examples available which you can just copy and paste.

For example look at the test: `TEST_P(MultiStackFixture, TestGithubExample) {` and how it uses the `DoClusteringTestImportedFromJson` method to run a test similar to what you need.

In CherrySimTester.cpp take a look at line `::testing::GTEST_FLAG(filter) = "*:-*_scheduled*:*_long*";` and modify it so that it will only run your custom Test. E.g. for the following example, use `::testing::GTEST_FLAG(filter) = "*MyOwnCustomTest*:-*_scheduled*:*_long*";` and it will only run your test and not all tests at once.

```
TEST(TestClustering, MyOwnCustomTest) {
    CherrySimTesterConfig testerConfig = CherrySimTester::CreateDefaultTesterConfiguration();
    SimConfiguration simConfig = CherrySimTester::CreateDefaultSimConfiguration();
    simConfig.terminalId = 0;
    testerConfig.verbose = true;
    simConfig.nodeConfigName.insert( { "prod_mesh_nrf52", 5 } );
    CherrySimTester tester = CherrySimTester(testerConfig, simConfig);
    tester.Start();

    tester.SimulateForGivenTime(5 * 1000);

    tester.SendTerminalCommand(1, "Blablabla");

    tester.SimulateUntilMessageReceived(10 * 1000, 1, "BlubBlub");
}
```

In this example, verbose is activated so you will see the terminal output of all nodes prefixed with time and nodeId for better understanding. You can open your browser at the provided URL (printed in the beginning) and you can see how the mesh forms. You can then take a look at the `DoClusteringTestImportedFromJson` method to see how you can import a json with your devices.

Marius

mabner1996, 2021-03-31:
[image]
Thank you for the kind reply

Now I tried the above test file that you show me in the tester so here is what happen
1. After the test run, the tester closes by itself, is it expected
2. Why there is command not found? Is the command not found making the error that closes cherrysim? Thanks

mariusheil, 2021-03-31:
Hello,

1) This is expected (normaly, not in your case ;-)) . The Google Test Framework runs all tests one after the other and then exits with exit code 0. This makes it possible to easily use it as part of a build pipeline.
2) Command not found is easily explained: I entered "Blablabla" which is not a valid FruityMesh command, valid commands would be stuff such as "action 0 status get_status" or "status", etc,....

You can also use e.g. SimulateUntilClusteringDone which will run the simulation until all nodes are sucessfully connected or SimulateForever which is good for manual testing.

Marius

mabner1996, 2021-03-31:
Noted, Thanks
Another thing I want to ask is that
So when I first build the simulator
I need to 
1. use cmake..FruityMesh -A Win32 in sibling directory to create the visualstudio solution
2, And then open to visualstudio solution and build the tester and runner

Do I need to do both of these for everytime I make a change in CherrysimTester.cpp and ClusteringTest.cpp?

Thanks

mabner1996, 2021-04-07:
Thank you very much.
Last question, in this simulation I actually want to simulate when the node is turned on and off and what happened to the connection when it happens.

So it goes like this:
1. Node 1 turned on at first, 5 minutes later Node 2 turned on, 5 minutes later we turn Node 3 on

As Usually I just upload json file which all the nodes automatically turned on in the simulator, how do I make this kind of sequence? Thanks

mariusheil, 2021-04-07:
Hi,

the simulator does not currently have an easy way to switch a node "off". You could disable discovery on a node, but I guess the easiest way to achieve what you want is:

- Modify the X and y position of your nodes that you want to have off and move them far away from the other nodes.
- at one point, move a node back to its original position and at the same time use the reset command

Marius

mabner1996, 2021-04-27:
thank you very much for your answer

closed 2021-04-27 by mabner1996