bluerange-io/bluerange-mesh #172: nordic usb dongle + fruitymesh

maagedmazyek, 2021-03-15:
Hi marius,

I think I oversee an option to enable fruitymesh with the nordic USB dongle. Whenever I upload the firmware, the USB dongle stays in DFU mode (and does **not** allow the connection via a serial terminal).

I found that the precompiled version `github_sink_usb_nrf52840_merged.hex`from you works out of the box with the USB dongle.

Any suggestions?

Thanks!
Maaged

mariusheil, 2021-03-15:
Hello,

we have not tested this Release with the nordic usb dongle using DFU. We have found this development board to be a bit fragile and hard to work with. We are using USB dongles of a different vendor internally. There are some problems with 1.8V as we are not activating this in the UICR.

Maybe this has something to do with it:
https://devzone.nordicsemi.com/nordic/short-range-guides/b/getting-started/posts/nrf52840-dongle-programming-tutorial

Good luck

mariusheil, 2021-06-23:
Hello, I'll close this as there has not been any more feedback. Feel free to reopen it.

closed 2021-06-23 by mariusheil