bluerange-io/bluerange-mesh #173: Ruuvi Measurements Missing

arctechllc, 2021-03-16:
I'm building out a wireless sensor network to track atmospheric readings.  One use case is custody chain for covid vaccines, and there are many others. 

I'm using RuuviTags as the sensor nodes, I'm using an esp32 M5Stack for a receiver, and I have a full backend I built on AWS using Timestream to house all readings. 

As it stands, the system works great so long as the tags are within range of an esp32 gateway device.  This is not always practical, however, and a mesh system would be very useful here so you can use only 1 gateway and let the tags mesh with one another (they are close to one another, but not necessarily to the gateway).

I flashed 2 new RuuviTags with the latest release Ruuvi merge firmware from your repository.  When I listen to either tag, I see 2 different advertisements over and over again, but neither one appears to change whatsoever so it's highly unlikely there is sensor data in them.

Moreover, when I use the RuuviStation app, it only updates once every 3 min if that, and the mesh itself doesn't appear to be working because I can only see sensors in the app which are within range. 

I'm using the latest version of RuuviTags (delivered yesterday).  Please see below the output from BLE listener:

---

17:53:19.360 MQT: tele/tasmota_454778/BLE = {"details":{"mac":"C31710AC92AF","RSSI":-97,"p":"0201061BFF4D02F00B0001690669060EE00200121C0401FFFF100000000000"}}
17:53:20.015 MQT: tele/tasmota_454778/BLE = {"details":{"mac":"C31710AC92AF","RSSI":-102,"p":"0201061BFF4D02F00B0001690669060EE00200121C0401FFFF100000000000"}}
17:53:26.009 MQT: tele/tasmota_454778/BLE = {"details":{"mac":"C31710AC92AF","RSSI":-99,"p":"0201061BFF4D02F00B0001690669060EE00200121C0401FFFF100000000000"}}
17:53:28.160 MQT: tele/tasmota_454778/BLE = {"details":{"mac":"C31710AC92AF","RSSI":-95,"p":"020106030312FE171612FE03000B0009961C29000000000100000000000000","0xfe12":"03000B0009961C29000000000100000000000000"}}
17:53:34.165 MQT: tele/tasmota_454778/BLE = {"details":{"mac":"C31710AC92AF","RSSI":-103,"p":"0201061BFF4D02F00B0001690669060EE00200121C0401FFFF100000000000"}}
17:53:34.461 MQT: tele/tasmota_454778/BLE = {"details":{"mac":"C31710AC92AF","RSSI":-99,"p":"020106030312FE171612FE03000B0009961C29000000000100000000000000","0xfe12":"03000B0009961C29000000000100000000000000"}}
17:53:36.461 MQT: tele/tasmota_454778/BLE = {"details":{"mac":"C31710AC92AF","RSSI":-99,"p":"020106030312FE171612FE03000B0009961C29000000000100000000000000","0xfe12":"03000B0009961C29000000000100000000000000"}}
17:53:37.819 MQT: tele/tasmota_454778/BLE = {"details":{"mac":"C31710AC92AF","RSSI":-98,"p":"0201061BFF4D02F00B0001690669060EE00200121C0401FFFF100000000000"}}
17:53:44.572 MQT: tele/tasmota_454778/BLE = {"details":{"mac":"C31710AC92AF","RSSI":-101,"p":"020106030312FE171612FE03000B0009961C29000000000100000000000000","0xfe12":"03000B0009961C29000000000100000000000000"}}
17:53:46.668 MQT: tele/tasmota_454778/BLE = {"details":{"mac":"C31710AC92AF","RSSI":-98,"p":"0201061BFF4D02F00B0001690669060EE00200121C0401FFFF100000000000"}}
17:53:47.619 MQT: tele/tasmota_454778/BLE = {"details":{"mac":"C31710AC92AF","RSSI":-101,"p":"020106030312FE171612FE03000B0009961C29000000000100000000000000","0xfe12":"03000B0009961C29000000000100000000000000"}}
17:53:47.972 MQT: tele/tasmota_454778/BLE = {"details":{"mac":"C31710AC92AF","RSSI":-99,"p":"020106030312FE171612FE03000B0009961C29000000000100000000000000","0xfe12":"03000B0009961C29000000000100000000000000"}}
17:53:48.273 MQT: tele/tasmota_454778/BLE = {"details":{"mac":"C31710AC92AF","RSSI":-98,"p":"020106030312FE171612FE03000B0009961C29000000000100000000000000","0xfe12":"03000B0009961C29000000000100000000000000"}}
17:53:48.992 MQT: tele/tasmota_454778/BLE = {"details":{"mac":"C31710AC92AF","RSSI":-97,"p":"0201061BFF4D02F00B0001690669060EE00200121C0401FFFF100000000000"}}
17:53:51.015 MQT: tele/tasmota_454778/BLE = {"details":{"mac":"C31710AC92AF","RSSI":-101,"p":"0201061BFF4D02F00B0001690669060EE00200121C0401FFFF100000000000"}}
17:53:51.310 MQT: tele/tasmota_454778/BLE = {"details":{"mac":"C31710AC92AF","RSSI":-98,"p":"0201061BFF4D02F00B0001690669060EE00200121C0401FFFF100000000000"}}

siretty, 2021-03-16:
Hi @arctechllc, in order to analyze your issue we will need some details on your setup.

- Your RuuviTags would be hardware revision B7.1, correct? You can find this revision number on the back of the circuit board in gold-plated copper next to the vias under the nRF chip. B7.1 is also the version I tested the firmware on.
- How many tags did you use for your test?
- How far away are the tags from each other and how far away were you from the nearest tag?
- How long did you wait before you recorded the advertisments?

The advertisments you did receive are not sensor data, but used by FruityMesh internally. The manufacturer id for Ruuvi Innovations Ltd. is [0x99, 0x04] (lsb), but the advertisments you see have manufacturer id [0x4D, 0x02] (lsb) which is M-Way Solutions GmbH.

Just going by your initial post and a hunch, it is likely that you are not close enough to the tags to receive the sensor data and potentially that the tags are not close enough to each other to establish a mesh.

Could you try to place at least two or three tags right in front of you and see if the mesh forms and the app picks up the sensor data from all of the tags?

arctechllc, 2021-03-16:
Hi @siretty

Thank you for the prompt response.

Indeed the version is b7.1

Im using 2 tags to test the mesh, and they are right next to each other in front of me about 30cm from my phone and 1m line of sight from the esp32

I tried to manually reset them a couple of times with a view that they may reestablish a link.

I will now reflash both and try again just in case something happened during flash process. I'm using Ruuvi shield to do this

Will update if anything changes my end

siretty, 2021-03-19:
Hello @arctechllc, just a quick heads up before you run into another issue yourself. There is a known issue with the RuuviWeatherModule in FruityMesh that leads to fairly high current consumption (drains batteries rather quickly at the moment). I'm actively working on that, just so you know.

arctechllc, 2021-03-19:
Hi @siretty

Many thanks for the information. I have not had time to dig into my setup but I did all the basic troubleshooting to no avail.

I'll wait to see the new commit. Good luck!

siretty, 2021-03-19:
Hey @arctechllc, attached is a debug build of the firmware.

It has the LED mode set to CONNECTIONS (so tags flashed with this firmware will blink red when not connected at all and if they are connected the number of green flashes indicates the number of connections). Also the terminal is accessible via JLinkRTTViewer, so you can see if any errors are generated. I also noticed that silly old me had set the transmission power of the sensor advertisments to -40dBm which may be related to you not receiving them (this was _really_ short range). This debug firmware has it set to +4dBm (which for me goes through at least two walls) (RuuviWeatherModule.cpp:110).

FYI: The sensor measurements are (on a single tag) generated in 30s intervals (this can be changed in the firmware though).

[2021-03-19-18-47_prod_ruuvi_weather_nrf52_merged_debug_with_leds_and_terminal.hex.zip](https://github.com/mwaylabs/fruitymesh/files/6173234/2021-03-19-18-47_prod_ruuvi_weather_nrf52_merged_debug_with_leds_and_terminal.hex.zip)

arctechllc, 2021-03-19:
@siretty Mate, you're awesome. I really appreciate this. Let me give it some tests and I'll come back on here

arctechllc, 2021-03-24:
Hi @siretty 

Thanks again for providing the debug firmware.  I can confirm the ruuvitags connect to one another as they both blink green once.  My sniffer picks up all the advertisements, but I cannot seem to pick up data from one by listening to the other.  Maybe I've misunderstood how this works?  Would the advertisement including data from other mesh nodes look different?

Here is a sample output from one of the 2 ruuvitags running the above firmware.  I bolded one of the lines that shows the readings from the tag whose mac I am sniffing, but I see nothing parallel for the other mac (E2:17:78:01:EF:F9)

Many thanks for your time in advance.

03:38:26.692 RSL: tele/tasmota_8C806C/BLE = {"details":{"mac":"F438B7A648DC","RSSI":-62,"p":"0201061BFF4D02F00B00011604BA00AC56020019200401FFFF100000000000"}}
03:38:27.042 RSL: tele/tasmota_8C806C/BLE = {"details":{"mac":"F438B7A648DC","RSSI":-66,"p":"0201061BFF4D02F00B00011604BA00AC56020019200401FFFF100000000000"}}
03:38:27.242 RSL: tele/tasmota_8C806C/BLE = {"details":{"mac":"F438B7A648DC","RSSI":-73,"p":"020106030312FE171612FE03000B0009E12929000000000100000000000000","0xfe12":"03000B0009E12929000000000100000000000000"}}
03:38:27.392 RSL: tele/tasmota_8C806C/BLE = {"details":{"mac":"F438B7A648DC","RSSI":-71,"p":"020106030312FE171612FE03000B0009E12929000000000100000000000000","0xfe12":"03000B0009E12929000000000100000000000000"}}
03:38:27.542 RSL: tele/tasmota_8C806C/BLE = {"details":{"mac":"F438B7A648DC","RSSI":-63,"p":"020106030312FE171612FE03000B0009E12929000000000100000000000000","0xfe12":"03000B0009E12929000000000100000000000000"}}
03:38:27.843 RSL: tele/tasmota_8C806C/BLE = {"details":{"mac":"F438B7A648DC","RSSI":-66,"p":"0201061BFF4D02F00B00011604BA00AC56020019200401FFFF100000000000"}}
03:38:28.193 RSL: tele/tasmota_8C806C/BLE = {"details":{"mac":"F438B7A648DC","RSSI":-73,"p":"0201061BFF4D02F00B00011604BA00AC56020019200401FFFF100000000000"}}
03:38:28.493 RSL: tele/tasmota_8C806C/BLE = {"details":{"mac":"F438B7A648DC","RSSI":-71,"p":"020106030312FE171612FE03000B0009E12929000000000100000000000000","0xfe12":"03000B0009E12929000000000100000000000000"}}
03:38:29.342 RSL: tele/tasmota_8C806C/BLE = {"details":{"mac":"F438B7A648DC","RSSI":-62,"p":"020106030312FE171612FE03000B0009E12929000000000100000000000000","0xfe12":"03000B0009E12929000000000100000000000000"}}
03:38:29.492 RSL: tele/tasmota_8C806C/BLE = {"details":{"mac":"F438B7A648DC","RSSI":-72,"p":"020106030312FE171612FE03000B0009E12929000000000100000000000000","0xfe12":"03000B0009E12929000000000100000000000000"}}
03:38:29.641 RSL: tele/tasmota_8C806C/BLE = {"details":{"mac":"F438B7A648DC","RSSI":-63,"p":"0201061BFF4D02F00B00011604BA00AC56020019200401FFFF100000000000"}}
03:38:29.792 RSL: tele/tasmota_8C806C/BLE = {"details":{"mac":"F438B7A648DC","RSSI":-72,"p":"0201061BFF4D02F00B00011604BA00AC56020019200401FFFF100000000000"}}
03:38:29.992 RSL: tele/tasmota_8C806C/BLE = {"details":{"mac":"F438B7A648DC","RSSI":-62,"p":"0201061BFF4D02F00B00011604BA00AC56020019200401FFFF100000000000"}}
03:38:30.293 RSL: tele/tasmota_8C806C/BLE = {"details":{"mac":"F438B7A648DC","RSSI":-62,"p":"0201061BFF4D02F00B00011604BA00AC56020019200401FFFF100000000000"}}
03:38:30.593 RSL: tele/tasmota_8C806C/BLE = {"details":{"mac":"F438B7A648DC","RSSI":-64,"p":"020106030312FE171612FE03000B0009E12929000000000100000000000000","0xfe12":"03000B0009E12929000000000100000000000000"}}
03:38:31.244 RSL: tele/tasmota_8C806C/BLE = {"details":{"mac":"F438B7A648DC","RSSI":-67,"p":"020106030312FE171612FE03000B0009E12929000000000100000000000000","0xfe12":"03000B0009E12929000000000100000000000000"}}
03:38:31.442 RSL: tele/tasmota_8C806C/BLE = {"details":{"mac":"F438B7A648DC","RSSI":-72,"p":"020106030312FE171612FE03000B0009E12929000000000100000000000000","0xfe12":"03000B0009E12929000000000100000000000000"}}
03:38:31.592 RSL: tele/tasmota_8C806C/BLE = {"details":{"mac":"F438B7A648DC","RSSI":-71,"p":"020106030312FE171612FE03000B0009E12929000000000100000000000000","0xfe12":"03000B0009E12929000000000100000000000000"}}
03:38:31.792 RSL: tele/tasmota_8C806C/BLE = {"details":{"mac":"F438B7A648DC","RSSI":-64,"p":"0201061BFF4D02F00B00011604BA00AC56020019200401FFFF100000000000"}}
03:38:31.892 RSL: tele/tasmota_8C806C/BLE = {"details":{"mac":"F438B7A648DC","RSSI":-71,"p":"0201061BFF4D02F00B00011604BA00AC56020019200401FFFF100000000000"}}
03:38:32.206 RSL: tele/tasmota_8C806C/BLE = {"details":{"mac":"F438B7A648DC","RSSI":-64,"p":"0201061BFF4D02F00B00011604BA00AC56020019200401FFFF100000000000"}}
03:38:33.171 RSL: tele/tasmota_8C806C/BLE = {"details":{"mac":"F438B7A648DC","RSSI":-63,"p":"0201061BFF4D02F00B00011604BA00AC56020019200401FFFF100000000000"}}
03:38:33.371 RSL: tele/tasmota_8C806C/BLE = {"details":{"mac":"F438B7A648DC","RSSI":-68,"p":"020106030312FE171612FE03000B0009E12929000000000100000000000000","0xfe12":"03000B0009E12929000000000100000000000000"}}
03:38:33.471 RSL: tele/tasmota_8C806C/BLE = {"details":{"mac":"F438B7A648DC","RSSI":-64,"p":"020106030312FE171612FE03000B0009E12929000000000100000000000000","0xfe12":"03000B0009E12929000000000100000000000000"}}
03:38:33.671 RSL: tele/tasmota_8C806C/BLE = {"details":{"mac":"F438B7A648DC","RSSI":-68,"p":"020106030312FE171612FE03000B0009E12929000000000100000000000000","0xfe12":"03000B0009E12929000000000100000000000000"}}
[... cut ...]
03:40:18.583 RSL: tele/tasmota_8C806C/BLE = {"details":{"mac":"F438B7A648DC","RSSI":-67,"p":"020106030312FE171612FE03000B0009E12929000000000100000000000000","0xfe12":"03000B0009E12929000000000100000000000000"}}
03:40:18.883 RSL: tele/tasmota_8C806C/BLE = {"details":{"mac":"F438B7A648DC","RSSI":-70,"p":"0201061BFF4D02F00B00011604BA00AC56020019200401FFFF100000000000"}}
03:40:19.233 RSL: tele/tasmota_8C806C/BLE = {"details":{"mac":"F438B7A648DC","RSSI":-70,"p":"0201061BFF4D02F00B00011604BA00AC56020019200401FFFF100000000000"}}
03:40:19.432 RSL: tele/tasmota_8C806C/BLE = {"details":{"mac":"F438B7A648DC","RSSI":-65,"p":"0201061BFF4D02F00B00011604BA00AC56020019200401FFFF100000000000"}}

siretty, 2021-04-22:
Hello @arctechllc,

I am very sorry for taking this long to find time to reply to your last comment. I hope you've been doing well.

In the meantime I have tried to reproduce your issue multiple times on different devices, but for me the connection is established and sensor data advertisments are generated as expected.

Would you be so kind as to show the command you used for flashing, and if you are able to access the log of at least one of the nodes available via JLinkRTTViewer when using the firmware from [my previous comment](https://github.com/mwaylabs/fruitymesh/issues/173#issuecomment-803012450)? There is a log message which is printed when sensor data is received from another node - which we can use to see if the problem is in the advertisment functionality, between the nodes or in the code reading and distributing the sensor data.

arctechllc, 2021-05-06:
Hi @siretty ,

Sorry it took me a while to come back. I just noticed the new release so I will set up a new demo network and try again. I think I may have been looking for the mesh messaging in the advertised payloads but this actually requires a direct connection to one of the nodes. Kind of an amateurish error on my end, but I'm still fairly new to this technology in general so I'm learning as I go. Do you have a preferred bluetooth sniffer app/hardware that you use to test whether the mesh is working or not? I will use JLink eventually but I'm waiting on some hardware to do that

mariusheil, 2021-06-23:
Hello, I'll close this as there has not been any more feedback. Feel free to reopen it.

closed 2021-06-23 by mariusheil