bluerange-io/bluerange-mesh #175: High power usage after SPI transaction (~3..8mA) with RuuviTag

Michaelhobo, 2021-05-04:
I used a variation of the prod_ruuvi_weather featureset, and noticed power usage was really high, around 4mA continuous. I removed the SPI transaction in the RuuviWeatherModule, and power usage went down to ~100uA.

I used a Joulescope to perform these measurements, and the moment the SPI transaction is called, the current stays high at ~3.8mA, even though the transaction only occurs every 30s.

I'm testing this on RuuviTags, and firmware release 1.0.520. Do you have any idea what this issue could be?

Michaelhobo, 2021-05-04:
I removed the call to `bme280_read_measurements` here: https://sourcegraph.com/github.com/mwaylabs/fruitymesh@master/-/blob/src/modules/RuuviWeatherModule.cpp#L962

siretty, 2021-05-04:
Hello @Michaelhobo, you might be running into a bug which will be fixed in the next GitHub release of FruityMesh. I attached a patch which includes the fix.

In case this does not help with the issue, how did you remove the call to the `bme280_read_mesaurements` function? Do you setup dummy data and are still advertising or did you deactivate the data processing in it's entirety?

I'm asking because the bug is triggered during processing of the sensor data (actually the known [anomaly 87](https://infocenter.nordicsemi.com/topic/errata_nRF52832_EngC/ERR/nRF52832/EngineeringC/latest/anomaly_832_87.html?cp=4_2_1_2_1_24) of certain nordic chips), not actually when doing the SPI transaction.

[FruityMesh_GitHub_Issue175_Mitigation_Anomaly87.patch.zip](https://github.com/mwaylabs/fruitymesh/files/6420124/FruityMesh_GitHub_Issue175_Mitigation_Anomaly87.patch.zip)

Michaelhobo, 2021-05-04:
This patch solved the issue, thanks for the patch. 

I could've named my issue a bit better, I removed "SPI transaction" and mentioned the actual function, for clarity.

siretty, 2021-05-05:
Thank you for the confirmation. I'll close this issue then.

closed 2021-05-05 by siretty