bluerange-io/bluerange-mesh #176: Module for Communication with Smartphones

nishinohi, 2021-05-23:
I have created a module that receives terminal commands from a smartphone via MeshAccessConnection and sends logs to the smartphone. I also created a simple smartphone application to communicate with that module.

A description of the app and module can be found at the following link.

* [FruityMeshAppUart](https://github.com/nishinohi/FruityMeshAppUart)

There are not a few limitations to the smartphone app, but if this module (AppUartModule) is useful for this PJ, I would like to send a pull request. I'm looking forward to your feedback.

mariusheil, 2021-05-31:
Hello,

this looks very nice. :-) I have put it on a list of topics for internal discussion. I would really like to merge it in, probably as an example together with a short Tutorial about what can be done with FruityMesh. We will however need to discuss internally if we can maintain the code in the future so that it does not break. I will be in touch again.

Marius

mariusheil, 2021-06-23:
Hi,

sorry for taking this long. We have discussed it internally. While the feature does not really fit into the core of FruityMesh, it could be a very nice showcase that other developers could work with. We would be fine to accept a pull request under a few conditions:
- The feature should be implemented as a vendor module (using a 4 byte VendorModuleId). This would only require minimal effort to change it, if corrently a 2 byte moduleId was used.
- The module could reside in the src/examples directory
- Some documentation in the form of a tutorial is available, that we can include as part of https://www.bluerange.io/docs/fruitymesh/Tutorials.html
- We we add a link to your repository on github where the source code of the android app is available. We cannot put the android source code in our repository as we will not be able to maintain it with our team, so an external link would probably the best option.

That's it :-)

Marius

nishinohi, 2021-06-24:
Thanks for the reply.
I am happy to be able to contribute because this PJ has given me a chance to learn more about BLE.
I will adjust it to meet the presented requirements and submit a pull request.

closed 2021-06-24 by nishinohi