bluerange-io/bluerange-mesh #179: Question: Zephyr OS compatible?

sieversb, 2021-06-24:
Hey everyone,

I was looking for a BLE Mesh algorithm which would allow to run all devices within the network on batteries. After I read a post about FruityMesh on the Nordic Website I got really excited since my desired application would need to operate fully on batteries! My question would be if this Fruitymesh project is integrable into my already existing Zephyr OS project or if this would require a lot of changes in the source code? Maybe someone as made some experiences with that already and is willing to share his/her approach on how to do so.

mariusheil, 2021-06-24:
Hi,
we do not have any Zephyr integration yet, but we will probably have it coming up on our Roadmap some time. I am not sure how your application is written but depending on how modular you wrote it, you might be able to put it on top of FruityMesh. You would probably only need to replace some calls where you build a connection, send data or advertise some stuff. 

Marius

sieversb, 2021-06-30:
Thank you for your response. I will see what I can do then.

closed 2021-07-16 by sieversb